google/aiyprojects-raspbian #2: Bulb flashing

pjrh, 2017-05-04:
Hi,

Really great kit!  I'm already having fun with it.

How can I stop the bulb flashing all the time when it's waiting for a button press?  The guide says I can do this "by modifying the configuration file in ~/voice-recognizer-raspi/config/status-led.ini.default", but I can't see how to make this work.

Thanks for your help,
Peter

ensonic, 2017-05-05:
Hi Peter,
the led flashing signals the kit starting up. When the voice-recognizer is configured to start automatically it will turn the led to an occasional pulse when it has started. You can use this command to make the service autostart:

    sudo systemctl enable voice-recognizer

(see https://aiyprojects.withgoogle.com/voice/#users-guide-3-2--manage-the-service)

If you just want to turn off the led, you can also run:

    echo "power-off" | sudo tee /tmp/status-led

The config file is for choosing the led. If you want to change the patterns associated with each action, you would need to do this in the sources:
https://github.com/google/aiyprojects-raspbian/blob/master/src/led.py#L126

pjrh, 2017-05-05:
Thanks, that has explained everything.  It's working exactly as I want now.

closed 2017-05-05 by pjrh