google/aiyprojects-raspbian #3: Openbox Syntax Error

drigz, 2017-05-05:
Some people have seen this error on first boot:

`One or more XML syntax errors were found while parsing the Openbox configuration files. See stdout for more information. The last error seen was in file "/home/pi/.config/openbox/lxde-pi-rc.xml" line 1, with message: Document is empty`

It can typically be resolved by reflashing the card.

This might be an upstream issue ([see this forum post](https://www.raspberrypi.org/forums/viewtopic.php?t=178680&p=1138406)). As far as I'm aware lxde-pi-rc.xml is the same in our image as in the upstream image.

bazwilliams, 2017-05-13:
I had this error, was able to recover a desktop but it was missing the icons used in the tutorials so reflashed. 

I booted first time with nothing plugged into HDMI port, I powered off then plugged in my display.

drigz, 2017-05-15:
More info in this bug report received by email:

I used image "aiyprojects-2017-05-03.img.xz" and couldn't get it to boot up to the desktop.

* It booted and showed this message "Openbox Syntax Error"  "One or more XML syntax errors were found while parsing the Openbox configuration files. See stdout for more information. The last error seen was in the file "/home/pi/.config/openbox/lxde-pi-rc.xml" line 1, with message: Document is empty."
* After clicking "Close" on that message, the screen stayed blank except for the mouse cursor that followed my mouse movements.
* However, I could right-click and get a menu and open a terminal window. And I verified that /home/pi/.config/openbox/lxde-pi-rc.xml was a file of 0 bytes dated 2017-05-03.
* I re-flashed the sdcard to the same image. (I'm positive it was the same, as Etcher was still running and I chose to flash with the "same image".)  This time it booted fine and the file /home/pi/.config/openbox/lxde-pi-rc.xml was a file of 24075 bytes dated 2017-05-03.
* What might account for this is that upon my initial boot, nothing showed up on the screen.  I think I gave it quite a long time to boot, but I did power down, ensure the HDMI cable was fully inserted, and then powered up and the screen showed the boot process.  Perhaps that power cycle damaged lxde-pi-rc.xml ??

I wonder if we could work around this by creating a backup of lxde-pi-rc.xml and trying to restore it on boot if the file is empty?

drigz, 2017-05-15:
I wonder if we could work around this by creating a backup of lxde-pi-rc.xml and trying to restore it on boot if the file is empty?

scugn1zz0, 2017-07-21:
I also have this issue, I initially booted without screen too

drigz, 2017-07-21:
Our current theory is that this happens if the user pulls the power cable out during the initialization process that happens on the first boot, which corrupts the SD card and requires reflashing. The reason for the link to the lack of monitor is that users don't know when they can safely remove power if they don't have a monitor connected.

@scugn1zz0, you should reflash your SD card - there's no other workaround at the moment.

scugn1zz0, 2017-07-23:
@drigz Thanks a lot for the information, I reflashed the SD card and it worked (this time i boot with HDMI in)

bwa11ace, 2017-07-26:
Can someone explain reflashing?

enetor, 2017-07-27:
@bwa11ace Basically it's just following the "Get the Voice Kit SD Image" section again and write the image to your SD card, see https://aiyprojects.withgoogle.com/voice/#assembly-guide-1-get-the-voice-kit-sd-image

wwwroth, 2017-08-10:
Found this issue via Google. I also recieved this error while booting a fresh SD install without a touchscreen plugged in and then plugging it in.

Rudysmama, 2017-11-12:
I received the error after I tried to change the config.txt to accept a different HDMI input.  I reflashed the image. Booted with a monitor and I was fine.  I was then able to set up the boot with VNC and SSH.  I have also installed the windows remote desktop versions to connect.  No additional issues.

Consoli0, 2017-12-30:
Mine booted perfectly the first time, then wouldnt display ANYTHING until I went into config.txt and changed #hdmi_safe=1 to hdmi_safe=1 (uncomment hdmi_safe=1)

Consoli0, 2017-12-30:
>drigz:
 _Our current theory is that this happens if the user pulls the power cable out during the initialization process that happens on the first boot, which corrupts the SD card and requires reflashing. The reason for the link to the lack of monitor is that users don't know when they can safely remove power if they don't have a monitor connected._
<

Tested that theory, its correct, the 'dirty bit' is set if the computer lost power while in the middle of processing.

navotera, 2018-02-03:
i believe reflash is not wise solution... 
i got this too... any better method to make me not to take reflashed step

drigz, 2018-02-05:
@navotera, if our theory is correct, this error occurs because the filesystem is corrupted when the user pulls the power cable from the Raspberry Pi while it is still running. In that case, reflashing is the most reliable solution.

If there are files on the Raspberry Pi that haven't been backed up, you can plug the SD card into a computer and copy them off before reflashing.

navotera, 2018-02-13:
i resolve this issues by removing default pi user and creating new user on whatever u name it.
dont forget to assign new user in root group roles

Camdude527, 2018-04-03:
Can someone please help with this, im having this issue but i have data and programs on the pi that i would hate to lose, is there a way to get them off?

drigz, 2018-04-06:
@Camdude527 You could plug the SD card into a computer to back up your data. And remember to keep regular backups of important data in future! eg with Google Drive / Dropbox / GitHub (for code)