google/aiyprojects-raspbian #6: Google Home android app does not detect the AIY projects home assistant

shivasiddharth, 2017-05-06:


drigz, 2017-05-08:
Thanks for your feedback!

Companion app support is in the [roadmap](https://developers.google.com/assistant/sdk/release-notes#coming_soon) and you can [sign up for updates](https://developers.google.com/assistant/sdk/overview#developer_preview_and_roadmap) if you're interested.

eduncan911, 2017-05-19:
+1 had me scratching my head as well.  Will sign up for updates.

ianholing, 2017-11-11:
+1 would love to link AIY to home app but it's not possible

rikkkurd, 2017-12-16:
would like to see it showing up in the google home app as well.

jpfk123, 2017-12-21:
The release notes of the latest SDK update seems to say it should connect to the app now but I cant find how. Anybody knows how to do it? I already created device model ID and followed steps from the embed project tutorials

shivasiddharth, 2017-12-21:
@jpfk123 Go into home control in the app and your device will be listed under virtual devices

jpfk123, 2017-12-22:
Thank you very much!! I was looking inside the google home app instead of the assistant settings. I thought I had done something wrong but its working now :-)

Matschi89, 2017-12-25:
@jpfk123
I'm totally new to the aiy kit and can't get it into the app. Can you help me?
How do I update the aiy kit to the new SDK and pair it with the app afterwards?
Thanks in advance
Cheers

t1m0thyj, 2017-12-26:
@Matschi89 Run these commands in a terminal in /home/pi/voice-recognizer-raspi (or wherever you have Assistant installed):
`git fetch origin pull/207/head:upgrade-assistant
git checkout upgrade-assistant
rm -rf ~/voice-recognizer-raspi/env
scripts/install-deps.sh
sudo systemctl restart voice-recognizer.service`
This will pull the updates from the PR submitted by @drigz, and update the Google Assistant library to version 0.1.0.
As @jpfk123 mentioned, it is not possible to see your AIY kit in the Google Home app, but once you have upgraded it should show up in the Google Assistant app on your phone (you should see "Voice Kit" listed under Settings -> Devices).

alainros, 2018-01-20:
I now see the AIY Kit in my Google Home app on my phone. But I can't assign it a room nor a Nickname.
It always responce with a error saying: " Can't update the setting. Check your connection."
Allmy devices are on the same Wifi Network, Phone, Google Home, Google mini and the AIY kit running on my Pi3

Any inside?

adharapuramkrish, 2018-02-07:
@t1m0thyj I followed your instructions but still no luck. I don't see "VoiceKit" on my Assistant app (neither on devices nor home control)