google/aiyprojects-raspbian #15: Configuring LED signals?

gleonard3, 2017-05-08:
What is the syntax for modifying the configuration file in ~/voice-recognizer-raspi/config/status-led.ini.default??  I need to turn off the "device ready" blinking, before my wife stabs me.

drigz, 2017-05-09:
Does ensonic's answer on #2 help?

gleonard3, 2017-05-11:
Yes, it does, thanks.  Changed "ready": "beacon-dark" to "ready": "off".
Sorry for the extra noise.

George Leonard,
(408) 372-7763

drigz, 2017-05-11:
Glad you've responded - I was terrified you'd fallen victim to mariticide. Have fun with the kit!

closed 2017-05-11 by drigz

LegoChicken, 2017-07-12:
I've tried modify the latest version of led.py to change "ready":
https://github.com/google/aiyprojects-raspbian/blob/master/src/led.py#L50
"ready": aiy.common.LED.LED_BEACON_DARK,
to
"ready": aiy.common.LED.LED_OFF
I then restarted the service:
sudo service voice-recognizer restart
(note: from memory so may not have got exact syntax right)
It seemed to make no difference. Any ideas what is wrong? I would really like to stop it pulsing when just sitting there.

drigz, 2017-07-12:
You need to restart the LED service:

`sudo service status-led restart`

LegoChicken, 2017-07-12:
Oh, I'll give that a try at home. Are there any other services associated with the project. I'll create a short script in to restart all of them.

LegoChicken, 2017-07-12:
I can see the list in:
https://github.com/google/aiyprojects-raspbian/tree/master/systemd
Would be useful to install a script (e.g. /usr/bin) that restarts everything, so those changing the code can experiment easily. I'll raise another issue for this.

addamm1, 2017-12-31:
Just found something in /AIY-voice-kit-python/src/aiy/_drivers. If you go into _status_ui.py, you can change what the actions do. I changed the ready one to "aiy.voicehat.LED.OFF" because i didn't like the led blinking all the time. Someone else probably found this, but just wanted to share.