google/aiyprojects-raspbian #23: No voice-recognizer.pid file when running main.py

jordandrako, 2017-05-10:
Following the HACKING.md file up to these steps:
``` shell
cd ~/voice-recognizer-raspi
source env/bin/activate
python3 src/main.py
```
This is the traceback I get.
``` shell
Traceback (most recent call last):
  File "src/main.py", line 302, in <module>
    main()
  File "src/main.py", line 139, in main
    create_pid_file(args.pid_file)
  File "src/main.py", line 105, in create_pid_file
    with open(file_name, 'w') as pid_file:
FileNotFoundError: [Errno 2] No such file or directory: '/run/user/1000/voice-recognizer.pid'
```

jordandrako, 2017-05-10:
This is the result of 
``` shell
cd ~/voice-recognizer-raspi
ls -l
id
```
``` shell
 ✝  ~/voice-recognizer-raspi   master  ls -l
total 64
drwxr-xr-x 2 pi pi  4096 May 10 09:11 checkpoints
drwxr-xr-x 2 pi pi  4096 May 10 09:11 config
-rw-r--r-- 1 pi pi   985 May 10 09:11 CONTRIBUTING.md
drwxr-xr-x 5 pi pi  4096 May 10 09:13 env
-rw-r--r-- 1 pi pi  1760 May 10 09:11 HACKING.md
-rw-r--r-- 1 pi pi 11358 May 10 09:11 LICENSE
-rw-r--r-- 1 pi pi   363 May 10 09:11 Makefile
drwxr-xr-x 3 pi pi  4096 May 10 09:11 po
-rw-r--r-- 1 pi pi   282 May 10 09:11 README.md
drwxr-xr-x 2 pi pi  4096 May 10 09:11 scripts
drwxr-xr-x 2 pi pi  4096 May 10 09:11 shortcuts
drwxr-xr-x 4 pi pi  4096 May 10 09:17 src
drwxr-xr-x 2 pi pi  4096 May 10 09:11 systemd
drwxr-xr-x 2 pi pi  4096 May 10 09:11 tests
 ✝  ~/voice-recognizer-raspi   master  id
uid=1000(pi) gid=1000(pi) groups=1000(pi),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),101(input),108(netdev),997(gpio),998(i2c),999(spi),1001(homeassistant)
```

drigz, 2017-05-10:
Thanks for the report! @jordandrako, what operating system are you using?

@ensonic, do you think we should fall back to /tmp or somewhere else if /run/user doesn't exist? The status-monitor would need to check for both too.

jordandrako, 2017-05-10:
I'm running a modified version of raspbian called [hassbian](https://github.com/home-assistant/pi-gen/releases/tag/v1.22).

jordandrako, 2017-05-10:
There is also a dependency issue when running main.py. It asks for PySocks, and using pip install PySocks in the venv makes it go away.

ensonic, 2017-05-10:
@drigz yeah, probably /tmp even though it is not meant for that.

@jordandrako can you let us know how your /run dir looks like?

jordandrako, 2017-05-10:
``` shell
total 44K
drwxr-xr-x 23 root        root         800 May 10 09:18 .
drwxr-xr-x 22 root        root        4.0K Apr 21 22:14 ..
drwxr-xr-x  2 avahi       avahi         80 May 10 09:17 avahi-daemon
-rw-r--r--  1 root        root           4 May 10 09:17 crond.pid
----------  1 root        root           0 May 10 09:17 crond.reboot
drwxr-xr-x  2 root        root          60 May 10 09:17 dbus
drwxr-xr-x  3 root        root         100 May 10 09:17 dhcpcd
-rw-r--r--  1 root        root           4 May 10 09:17 dhcpcd.pid
srw-rw----  1 root        root           0 May 10 09:17 dhcpcd.sock
srw-rw-rw-  1 root        root           0 May 10 09:17 dhcpcd.unpriv.sock
drwxr-xr-x  2 root        root          60 May 10 09:17 dnsmasq
drwxr-x---  2 Debian-exim Debian-exim   60 May 10 09:18 exim4
prw-------  1 root        root           0 May 10 09:17 hogsuspend
lrwxrwxrwx  1 root        root          25 May 10 09:17 initctl -> /run/systemd/initctl/fifo
drwxrwxrwt  4 root        root         100 May 10 09:17 lock
drwxr-xr-x  3 root        root          60 May 10 09:17 log
-rw-r--r--  1 root        root           4 May 10 09:17 mosquitto.pid
drwxr-xr-x  2 root        root          60 May 10 09:17 mount
drwxr-xr-x  2 mysql       root          80 May 10 09:17 mysqld
drwxr-xr-x  2 root        netdev        80 May 10 09:17 network
-rw-r--r--  1 root        root           3 May 10 09:17 ntpd.pid
drwxr-xr-x  2 root        root          40 May 10 09:17 plymouth
drwxr-xr-x  4 root        root          80 May 10 09:17 resolvconf
-rw-r--r--  1 root        root           4 May 10 09:17 rngd.pid
-rw-r--r--  1 root        root           4 May 10 09:17 rsyslogd.pid
drwxr-xr-x  2 root        root          60 May 10 09:17 sendsigs.omit.d
lrwxrwxrwx  1 root        root           8 May 10 09:17 shm -> /dev/shm
drwxr-xr-x  2 root        root          60 May 10 09:17 snapclient
drwxr-xr-x  2 root        root          60 May 10 09:17 snapserver
drwxr-xr-x  2 root        root          40 May 10 09:17 sshd
-rw-r--r--  1 root        root           4 May 10 09:17 sshd.pid
drwxr-xr-x 15 root        root         360 May 10 09:17 systemd
-rw-r--r--  1 root        root           4 May 10 09:17 thd.pid
srwxr-xr-x  1 root        root           0 May 10 09:17 thd.socket
drwxr-xr-x  2 root        root          60 Dec 31  1969 tmpfiles.d
drwxr-xr-x  7 root        root         160 May 10 09:17 udev
drwxr-xr-x  2 root        root          40 May 10 09:17 user
-rw-rw-r--  1 root        utmp        1.5K May 10 09:18 utmp
drwxr-x---  2 root        netdev        60 May 10 09:17 wpa_supplicant
-rw-r--r--  1 root        root           4 May 10 09:17 wpa_supplicant.wlan0.pid
```

ensonic, 2017-05-10:
Ahh, so there is /run/user. Could you please also run:

  ls -al  /run/user

and 

  id

Thanks!

jordandrako, 2017-05-10:
Nothing there
``` shell
total 0
drwxr-xr-x  2 root root  40 May 10 09:17 .
drwxr-xr-x 23 root root 800 May 10 09:18 ..
```

ensonic, 2017-05-10:
Okay, it is just missing the subdir for the user-id, which only root could create :/

jordandrako, 2017-05-10:
I can create the directory myself to get it to run, does it need to be owned by pi?

jordandrako, 2017-05-10:
I created the 1000 directory under /run/user and made pi the owner. Now running src/main.py gives this traceback:
``` shell
Traceback (most recent call last):
  File "src/main.py", line 302, in <module>
    main()
  File "src/main.py", line 159, in main
    do_recognition(args, recorder, recognizer, player)
  File "src/main.py", line 175, in do_recognition
    import triggers.gpio
  File "/home/pi/voice-recognizer-raspi/src/triggers/gpio.py", line 19, in <module>
    import RPi.GPIO as GPIO
ImportError: No module named 'RPi'
```
I then ran
``` shell
pip install Rpi.GPIO
```
Now the result of src/main.py is:
``` shell
[2017-05-10 10:48:41,639] INFO:audio:started recording
```
This is the desired behavior, yes?

jordandrako, 2017-05-10:
Hmm. I also had to install Rpi.GPIO outside the for the status-led.service.
``` shell
sudo python3 pip install Rpi.GPIO
``` 

After that I rebooted, which I found out deleted the /run/user/1000 directory, so I had recreate, and re chown that. Then I restarted all the services and it seems to work.

PR #50 (2017-05-18): Use the right location for the pidfile
We try /run/user/\<uid\>/ and /tmp/, in that order. This should ensure
functionality across a wide range of distros.

Fixes #23.
closed 2017-05-18 by drigz