google/aiyprojects-raspbian #24: Instructions to use your own Audio Input and Output with provided OS image

diggy106, 2017-05-10:
I am trying to use this project using my own bluetooth speaker / USB microphone. When running src/main.py from a termainal I am able to get things working after simply changing Pulseaudio sources to use my mic and my speaker. When voice-recoginzer runs as a daemon, it does not properly detect the 'default' alsa device the same way when I run aplay or arecord. I have gotten the mic to work by passing in the HW device with the -I flag, but I am unable to get my bluetooth speaker to work when voice-recognizer is running as a daemon. `aplay -l` only shows the Voice HAT as an audio output device.

It would be great if there were some instructions on how to BYO - mic / speaker using the provided OS image.

drigz, 2017-05-11:
The voice recognizer should work with any audio setup if ALSA is configured correctly. The ALSA configuration can be difficult, though, and depends on your hardware setup (eg USB soundcard, onboard audio, HDMI audio, bluetooth speaker). We don't plan to provide instructions for setups other than the AIY Projects Voice Kit.

#7 has some more info. Maybe you could post your ALSA config here if you get it to work, so others can use it as a guide?

closed 2017-05-11 by drigz

salmanfarisvp, 2017-05-11:
@diggy106 Can you please Explain how to setup Speaker and USB mic with this AIY Image .
Thanks in Advance

diggy106, 2017-05-11:
@drigz thanks for following up. I have the setup working when running it in the "dev console" but when it runs as a daemon I am getting errors using the exact same arecord and aplay commands that the voice recognizer is using. I simply add the bluetooth speaker/usb mic as the default Pulseaudio sink/source and the dev console works as well as the check_audio script. 

Is there anything that the daemon does differently which would cause this? I will do some more digging tonight.

drigz, 2017-05-11:
Ah, I see. You can inspect systemd/voice-recognizer.service to see how it's started. A couple of possible differences:

- Are you running as user `pi` from the terminal?
- Are you starting the service with `sudo systemctl start voice-recognizer` or by enabling it and restarting the Pi?
- Are you passing the `-I/--input-device` or `-O/--output-device` parameters which wouldn't be used by the service?
- When it starts as a service, what errors do you see with `sudo journalctl -n 20 -u voice-recognizer`?

Also, the voice-recognizer doesn't use PulseAudio, so you need to make sure the default ALSA device is correct. However, if check_audio is working, then it should be correct.

diggy106, 2017-05-11:
Are you running as user pi from the terminal? - Yes
Are you starting the service with sudo systemctl start voice-recognizer or by enabling it and restarting the Pi? - Running `sudo systemctl start voice-recognizer`
Are you passing the -I/--input-device or -O/--output-device parameters which wouldn't be used by the service? - No, simply running `src/main.py` works in dev console and I checked the systemd file and it uses no flags as you mentioned.
When it starts as a service, what errors do you see with sudo journalctl -n 20 -u voice-recognizer? I will get you that output.

diggy106, 2017-05-12:
This gist (https://gist.github.com/diggy106/f71e55dd7db92de18dbef5baa838ccb6) has some output from the command you gave me and /var/log/syslog. I added a line in there to also log the aplay or arecord command.

I've also attached videos to show that check_audio.py and src/main.py works with no flags, but the daemon fails to play audio on the bluetooth speaker. I can see that it is picking up my microphone.

check_audio.py - https://youtu.be/2D2UsFHHMkE
src/main.py - https://youtu.be/Ab-Yi2_1cUQ
daemon - https://youtu.be/5fWBv0U7-h8

ensonic, 2017-05-12:
This article shows how to modify the alsa config:
http://www.androidauthority.com/build-google-assistant-raspberry-pi-770296/

drigz, 2017-05-12:
@diggy106 Thanks for the videos, I have a much better idea what's going on now (but still no idea what's causing it!)

You could try running `aplay -v` on the console to show what device ALSA is using there, then patching audio.py to change `'-q'` to `'-v'` to see that output when the voice-recognizer is running as a daemon.

I'm surprised that `enabling/disabling audio amp` is in the syslog, since you're not using the Voice HAT. Have you edited your /boot/config.txt? If you have any of the lines `dtoverlay=i2s-mmap`, `dtoverlay=googlevoicehat-soundcard` or `dtparam=i2s=on` in there, you could try removing them and seeing if it fixes your problem.

diggy106, 2017-05-13:
I did not change /boot/config.txt myself. I do see the lines you mentioned in that file. Seems to come from https://github.com/google/aiyprojects-raspbian/blob/master/scripts/configure-driver.sh which I assume was run as part of the image. 

```
pi@raspberrypi:~ $ sudo grep dtoverlay=i2s-mmap /boot/config.txt 
dtoverlay=i2s-mmap
pi@raspberrypi:~ $ sudo grep dtoverlay=googlevoicehat-soundcard /boot/config.txt 
dtoverlay=googlevoicehat-soundcard
pi@raspberrypi:~ $ sudo grep dtparam=i2s=on /boot/config.txt 
dtparam=i2s=on
```

Below is the output from aplay -v.
```
pi@raspberrypi:~ $ aplay -v /usr/share/sounds/alsa/Front_Center.wav
Playing WAVE '/usr/share/sounds/alsa/Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
ALSA <-> PulseAudio PCM I/O Plugin
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 1
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 24000
  period_size  : 6000
  period_time  : 125000
  tstamp_mode  : NONE
  period_step  : 1
  avail_min    : 6000
  period_event : 0
  start_threshold  : 24000
  stop_threshold   : 24000
  silence_threshold: 0
  silence_size : 0
  boundary     : 1572864000
```

Here is syslog from the daemon with aplay -v
```
May 13 14:02:06 raspberrypi python3[1658]: [2017-05-13 14:02:06,059] INFO:main:listening...
May 13 14:02:06 raspberrypi python3[1658]: [2017-05-13 14:02:06,062] INFO:main:recognizing...
May 13 14:02:09 raspberrypi python3[1658]: [2017-05-13 14:02:09,364] INFO:speech:event_type: 1
May 13 14:02:09 raspberrypi python3[1658]: [2017-05-13 14:02:09,370] INFO:main:thinking...
May 13 14:02:09 raspberrypi python3[1658]: [2017-05-13 14:02:09,373] INFO:speech:transcript: test
May 13 14:02:09 raspberrypi python3[1658]: [2017-05-13 14:02:09,590] INFO:main:Playing 5.1630 seconds of audio...
May 13 14:02:09 raspberrypi python3[1658]: [2017-05-13 14:02:09,591] INFO:audio:aplay: self._cmd: ['aplay', '-v', '-t', 'raw', '-D', 'default', '-c', '1', '-f', 's16', '-r', '16000']
May 13 14:02:09 raspberrypi python3[1658]: Playing raw data 'stdin' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono
May 13 14:02:09 raspberrypi python3[1658]: Plug PCM: Rate conversion PCM (48000, sformat=S16_LE)
May 13 14:02:09 raspberrypi python3[1658]: Converter: libspeex (builtin)
May 13 14:02:09 raspberrypi python3[1658]: Protocol version: 10002
May 13 14:02:09 raspberrypi python3[1658]: Its setup is:
May 13 14:02:09 raspberrypi python3[1658]: stream       : PLAYBACK
May 13 14:02:09 raspberrypi python3[1658]: access       : RW_INTERLEAVED
May 13 14:02:09 raspberrypi python3[1658]: format       : S16_LE
May 13 14:02:09 raspberrypi python3[1658]: subformat    : STD
May 13 14:02:09 raspberrypi python3[1658]: channels     : 1
May 13 14:02:09 raspberrypi python3[1658]: rate         : 16000
May 13 14:02:09 raspberrypi python3[1658]: exact rate   : 16000 (16000/1)
May 13 14:02:09 raspberrypi python3[1658]: msbits       : 16
May 13 14:02:09 raspberrypi python3[1658]: buffer_size  : 8000
May 13 14:02:09 raspberrypi python3[1658]: period_size  : 2000
May 13 14:02:09 raspberrypi python3[1658]: period_time  : 125000
May 13 14:02:09 raspberrypi kernel: [  416.025835] voicehat-codec voicehat-codec: Enabling audio amp...
May 13 14:02:09 raspberrypi python3[1658]: tstamp_mode  : NONE
May 13 14:02:09 raspberrypi python3[1658]: period_step  : 1
May 13 14:02:09 raspberrypi python3[1658]: avail_min    : 2000
May 13 14:02:09 raspberrypi python3[1658]: period_event : 0
May 13 14:02:09 raspberrypi python3[1658]: start_threshold  : 8000
[... 33 lines cut ...]
May 13 14:02:09 raspberrypi python3[1658]: format       : S32_LE
May 13 14:02:09 raspberrypi python3[1658]: subformat    : STD
May 13 14:02:09 raspberrypi python3[1658]: channels     : 2
May 13 14:02:09 raspberrypi python3[1658]: rate         : 48000
May 13 14:02:09 raspberrypi python3[1658]: exact rate   : 48000 (48000/1)
May 13 14:02:09 raspberrypi python3[1658]: msbits       : 32
May 13 14:02:09 raspberrypi python3[1658]: buffer_size  : 24000
May 13 14:02:09 raspberrypi python3[1658]: period_size  : 6000
May 13 14:02:09 raspberrypi python3[1658]: period_time  : 125000
May 13 14:02:09 raspberrypi python3[1658]: tstamp_mode  : NONE
May 13 14:02:09 raspberrypi python3[1658]: period_step  : 1
May 13 14:02:09 raspberrypi python3[1658]: avail_min    : 6000
May 13 14:02:09 raspberrypi python3[1658]: period_event : 0
May 13 14:02:09 raspberrypi python3[1658]: start_threshold  : 24000
May 13 14:02:09 raspberrypi python3[1658]: stop_threshold   : 24000
May 13 14:02:09 raspberrypi python3[1658]: silence_threshold: 0
May 13 14:02:09 raspberrypi python3[1658]: silence_size : 0
May 13 14:02:09 raspberrypi python3[1658]: boundary     : 1572864000
May 13 14:02:09 raspberrypi python3[1658]: appl_ptr     : 0
May 13 14:02:09 raspberrypi python3[1658]: hw_ptr       : 0
May 13 14:02:14 raspberrypi kernel: [  421.275435] voicehat-codec voicehat-codec: Disabling audio amp...
May 13 14:02:14 raspberrypi python3[1658]: [2017-05-13 14:02:14,911] INFO:main:ready...
```

I'll try removing the lines you've mentioned and let you know.

diggy106, 2017-05-13:
I was finally able to get it working with the daemon by following some of the instructions in that Android Authority article linked above. I am going to stick with the 3.5mm setup as it is more reliable than the bluetooth connection and don't have to worry about auto-connecting on startup.

/etc/asound.conf
```
pcm.!default {
  type asym
  capture.pcm "mic"
  playback.pcm "speaker"
}
pcm.mic {
  type plug
  slave {
    pcm "hw:1,0"
  }
}
pcm.speaker {
  type plug
  slave {
    pcm "hw:0,0"
  }
}
```

Added this to the bottom of my /boot/config.txt
```
# Enable audio (loads snd_bcm2835)
dtparam=audio=on
#dtoverlay=i2s-mmap
#dtoverlay=googlevoicehat-soundcard
```

syslog
```
May 13 14:32:47 raspberrypi python3[1244]: Plug PCM: Hardware PCM card 0 'bcm2835 ALSA' device 0 subdevice 0
May 13 14:32:47 raspberrypi python3[1244]: Its setup is:
May 13 14:32:47 raspberrypi python3[1244]: stream       : PLAYBACK
May 13 14:32:47 raspberrypi python3[1244]: access       : RW_INTERLEAVED
May 13 14:32:47 raspberrypi python3[1244]: format       : S16_LE
May 13 14:32:47 raspberrypi python3[1244]: subformat    : STD
May 13 14:32:47 raspberrypi python3[1244]: channels     : 1
May 13 14:32:47 raspberrypi python3[1244]: rate         : 16000
May 13 14:32:47 raspberrypi python3[1244]: exact rate   : 16000 (16000/1)
May 13 14:32:47 raspberrypi python3[1244]: msbits       : 16
May 13 14:32:47 raspberrypi python3[1244]: buffer_size  : 8000
May 13 14:32:47 raspberrypi python3[1244]: period_size  : 2000
May 13 14:32:47 raspberrypi python3[1244]: period_time  : 125000
May 13 14:32:47 raspberrypi python3[1244]: tstamp_mode  : NONE
May 13 14:32:47 raspberrypi python3[1244]: period_step  : 1
May 13 14:32:47 raspberrypi python3[1244]: avail_min    : 2000
May 13 14:32:47 raspberrypi python3[1244]: period_event : 0
May 13 14:32:47 raspberrypi python3[1244]: start_threshold  : 8000
May 13 14:32:47 raspberrypi python3[1244]: stop_threshold   : 8000
May 13 14:32:47 raspberrypi python3[1244]: silence_threshold: 0
May 13 14:32:47 raspberrypi python3[1244]: silence_size : 0
May 13 14:32:47 raspberrypi python3[1244]: boundary     : 2097152000
May 13 14:32:47 raspberrypi python3[1244]: appl_ptr     : 0
May 13 14:32:47 raspberrypi python3[1244]: hw_ptr       : 0
May 13 14:32:49 raspberrypi python3[1244]: [2017-05-13 14:32:49,227] INFO:main:ready...
```

devilmind, 2017-08-17:
Hi,
working fine what you have suggested here but soon after reboot on my /home/pi/.asoundrc are appendend those line:

```
pcm.!default {
        type hw
        card 1
}

ctl.!default {
        type hw
        card 1
}
```

what is appending those line on reboot?
I need to delet them to get it worked again.
Can you help me please?

Thanks a lot.

devilmind, 2017-08-17:
I have solved myself using rapi-config,
If I force to use the jack option the .asoundrc file will not modified on reboot and the config remains in place.