google/aiyprojects-raspbian #29: Unable to run scripts on startup

shivasiddharth, 2017-05-11:
I set up a wake word using snowboy. But when i tried to start the script on boot it does not seem to do anything. Following are the things that i tried.
1. setup crontab.
2. included the script in rc.local
3. tried adding it in init.d 
But all these seem to do nothing.
Any help is appreciable. TIA

ensonic, 2017-05-11:
Nice job!
FYI, we're including ready made systemd startup scripts:
https://aiyprojects.withgoogle.com/voice/#users-guide-3-3--led-status-codes
If it works when you start it manually, all you need to do is:

    sudo systemctl enable voice-recognizer

Lets us know if that fixes it and please share your work!

shivasiddharth, 2017-05-11:
What i actually meant is, the voice recognizer is working fine on boot but the script to start the snowboy is not running on start-up. If i could get the snowboy script to start on boot, i could complete the documentation and share them here.

ensonic, 2017-05-11:
You can take some inspiration from
https://github.com/google/aiyprojects-raspbian/blob/master/systemd/voice-recognizer.service
and make one for snowboy. Here is the script that installs the services:
https://github.com/google/aiyprojects-raspbian/blob/master/scripts/install-services.sh

shivasiddharth, 2017-05-11:
Thanks for the pointers Stephan.

ensonic, 2017-05-12:
Please reopen if you still have issues.

closed 2017-05-12 by ensonic