google/aiyprojects-raspbian #43: StatusCode Unauthenticated. Request had invalid authentication credentials

VR6Pete, 2017-05-16:
After a period of random time, the service stops working and I have to restart the service to get it to start working again.

Screenshot of output attached below;

[image]

Unsure what could be causing that. please let me know if you require any more information.

Thanks ,

Pete

drigz, 2017-05-17:
Thanks for the report! Are you using the Assistant API or the Cloud Speech API?

How long is the period of time approximately? eg is it minutes or hours or days? We'll try to reproduce it here.

VR6Pete, 2017-05-17:
Hi There @drigz I have using Assistant API. 

I would guess at ~ 1 hour - 2 hours I will be able to extract more approximate information when I take a look at it later.

Cheers

Pete

drigz, 2017-05-25:
@VR6Pete, have you figured anything out about how to reproduce this?

I saw this two or three times while doing a demo a few days ago, but when I tried to reproduce it on the next day, I couldn't. The cube has been running about 24 hours and still works...

VR6Pete, 2017-05-25:
That was the same with me but unfortunately it's not happened again since the bug report.

closed 2018-06-29 by dmitriykovalev