google/aiyprojects-raspbian #45: enable actions via config file

ensonic, 2017-05-17:
We're gettting PRs for actions like:

- shutdown and reboot commands: https://github.com/google/aiyprojects-raspbian/pull/30
- hue lights: https://github.com/google/aiyprojects-raspbian/pull/33

Some of the actions need parameters (hue) and some actions are potentially dangerous, so that not everyone wants them. It would be nice to have a way to enable/disable them and also pass options for them.

VR6Pete, 2017-05-18:
Great idea

sheridat, 2017-05-30:
Excellent - exactly what is needed to stop src/action.py becoming a nightmare.

drigz, 2017-10-18:
Since the new direction of this project is to provide a support library (along with demos of how to use it) rather than a full-featured solution, I'm going to close this issue.

closed 2017-10-18 by drigz