google/aiyprojects-raspbian #54: LED not lighting up - installed on own raspberry build

VR6Pete, 2017-05-19:
Hi there,

I have setup as per HACKING.MD and the LED button doesn't light up, nor does voice recognizer start from systemctl on boot up.

If I start from main.py - the console says: info:audio:started recording but doesn't actually appear to be accepting any commands and LED still does not light up.

If I insert the card with the AIY image already loaded it works fine, LED lights up and everything works as expected. 

Are there some steps missing ?

What can I do to troubleshoot ?

drigz, 2017-05-19:
Can you post the output of this command?

`sudo journalctl -u status-led -n 20`

VR6Pete, 2017-05-19:
-- Logs begin at Fri 2017-05-19 13:40:44 UTC, end at Fri 2017-05-19 15:08:38 UTC. --
May 19 13:40:50 raspberrypi systemd[1]: status-led.service: main process exited, code=exited, status=1/FAILUR
May 19 13:40:50 raspberrypi systemd[1]: Unit status-led.service entered failed state.
May 19 13:40:50 raspberrypi systemd[1]: status-led.service holdoff time over, scheduling restart.
May 19 13:40:50 raspberrypi systemd[1]: Stopping status led service...
May 19 13:40:50 raspberrypi systemd[1]: Starting status led service...
May 19 13:40:50 raspberrypi systemd[1]: Started status led service.
May 19 13:40:51 raspberrypi bash[1452]: Traceback (most recent call last):
May 19 13:40:51 raspberrypi bash[1452]: File "src/led.py", line 160, in <module>
May 19 13:40:51 raspberrypi bash[1452]: main()
May 19 13:40:51 raspberrypi bash[1452]: File "src/led.py", line 117, in main
May 19 13:40:51 raspberrypi bash[1452]: import configargparse
May 19 13:40:51 raspberrypi bash[1452]: ImportError: No module named 'configargparse'
May 19 13:40:51 raspberrypi systemd[1]: status-led.service: main process exited, code=exited, status=1/FAILUR
May 19 13:40:51 raspberrypi systemd[1]: Unit status-led.service entered failed state.
May 19 13:40:51 raspberrypi systemd[1]: status-led.service holdoff time over, scheduling restart.
May 19 13:40:51 raspberrypi systemd[1]: Stopping status led service...
May 19 13:40:51 raspberrypi systemd[1]: Starting status led service...
May 19 13:40:51 raspberrypi systemd[1]: status-led.service start request repeated too quickly, refusing to st
May 19 13:40:51 raspberrypi systemd[1]: Failed to start status led service.
May 19 13:40:51 raspberrypi systemd[1]: Unit status-led.service entered failed state.

drigz, 2017-05-19:
Once #55 has been merged, you'll need to run:
```
git pull
sudo scripts/install-services.sh
```

VR6Pete, 2017-05-19:
Excellent - seems to be working ok now - I manually made the change to the service file and seems to be OK now.

drigz, 2017-05-19:
Great - thanks for reporting the issue!

PR #55 (2017-05-22): Use virtualenv for running the status-led service
This fixes #54, caused by breakage introduced when I moved
configargparse from a system install (which required the stretch
package) to the virtualenv (#16).
commit 4ca5dbbc (2017-05-22): Use virtualenv for running the status-led service (#55)

This fixes #54, caused by breakage introduced when I moved
configargparse from a system install (which required the stretch
package) to the virtualenv (#16).
closed 2017-05-22 by drigz