google/aiyprojects-raspbian #56: pip install --upgrade google-assistant-sdk broke main.py

VR6Pete, 2017-05-19:
Hi,

I have performed pip install --upgrade google-assistant-sdk to get the latest SDK for hotword support etc...

However main.py will not run anymore.

file src/main.py line 26, in <module>
from googlesamples.assistant import auth_helpers
ImportError: cannot import name 'auth_helpers'

Thanks.

Pete

GregHolmes, 2017-05-19:
Unfortunately I get the same issue as above. Although main.py never ran correctly for me. I had installed the latest Raspbian (Not the recommended images) and followed the HACKING.md file.

shivasiddharth, 2017-05-22:
Did you try this ? `python -m pip install --upgrade https://github.com/googlesamples/assistant-sdk-python/releases/download/0.3.0/google_assistant_library-0.0.2-py2.py3-none-linux_armv7l.whl`

Rockman18, 2017-05-22:
Of course yes but in the new google-assistant-library, they have removed the auth_helpers library used by aiyproject to generate the credentials. So it's necessary to recode the auth mechanism in aiyproject

shivasiddharth, 2017-05-22:
i thought i was alone on this !

GregHolmes, 2017-05-22:
So the idea is to install an older version that contains the auth_helpers library?

shivasiddharth, 2017-05-22:
I replaced the main.py in AIY projects with  [this](https://github.com/googlesamples/assistant-sdk-python/blob/master/google-assistant-sdk/googlesamples/assistant/library/hotword.py) renamed it as main.py and it started working, but you cannot control the GPIOs or volume like the AIY kit.

proppy, 2017-05-26:
Related #64 #67

eduncan911, 2017-05-27:
I had to:

    $  cp -r ~/assistant-sdk-python/googlesamples ~/voice-recognizer-raspi/src/

Note: i am in the middle up updating my Voice Kit ATM.  It runs now, after I did a `git pull` on the voice kit repo, and I had `source env/bin/activate` and then `env/bin/pip install phue`.

Now `src/main.py` runs!

shivasiddharth, 2017-05-27:
@eduncan911 , The requirements.txt still has the old sdk listed. So even if you reinstall voice recognizer, it will install the old SDK. Please correct me if i am wrong.

proppy, 2017-06-06:
This should be fixed now that #67 is closed.

closed 2017-06-27 by drigz