google/aiyprojects-raspbian #57: 'AssistantSpeechRequest' error

wrumble, 2017-05-19:
I keep getting the following error after a task has finished performing 

Traceback (most recent call last):
  File "/usr/lib/python3.4/threading.py", line 920, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.4/threading.py", line 868, in run
    self._target(*self._args, **self._kwargs)
  File "src/main.py", line 286, in _recognize
    if self.recognizer.dialog_follow_on:
AttributeError: 'AssistantSpeechRequest' object has no attribute 'dialog_follow_on'

It does it after any task but namely after i added the bbc radio stations addition found in this forum and i cant see why, i succesfully added the hotword, google play and youtube additions without any problems but now this has happened. Any recommendations how i can go about fixing it?

drigz, 2017-05-22:
Do you have any local modifications to main.py or speech.py? You can check by running `git status` or `git diff origin/master`.

25b55e1f added `dialog_follow_on` to both files, so if both are up-to-date there should be no problem.

wrumble, 2017-05-22:
Hi, thanks for replying, i had added a bit to main to add hotword support. Ended up re installing the image and going through the hotword steps again, which worked. Now struggling to get it to output through a bluetooth speaker. Nothing seems to work :( Thanks for replying

drigz, 2017-05-22:
OK, thanks - I'll close the issue.

Good luck with the bluetooth speaker - make sure to share if you figure out how to do it!

closed 2017-05-22 by drigz