google/aiyprojects-raspbian #58: errors in pinout documentation

RogueM, 2017-05-20:
I couldn't find a contact on the main site, so posting this here...

there are a number of errors in the documentation of the header breakouts at:
https://aiyprojects.withgoogle.com/voice/#makers-guide-4-1--connecting-additional-sensors

in https://aiyprojects.withgoogle.com/static/images/aiy-projects-voice/makers/sensors.png
MISO should correspond to bcm10 (or possibly it should be MOSI, depends on board routing, don't have one to check - but either way, one of these 2 is incorrect).

also in the detailed pinout below, under SPI, there is a mix of physical and bcm and physical notations. It should be `10, 9, 11, 8, 7` , not `10, 9, 11, 24, 26`.

I hope you can pass on this report to whoever is responsible for the aiyprojects.withgoogle.com site.

cheers,
HAL

drigz, 2017-05-22:
Thanks, I've passed this on. I'll update once we've fixed the site.

drigz, 2017-05-30:
Done - thanks again for reporting the error!

closed 2017-05-30 by drigz