google/aiyprojects-raspbian #59: Voice Trigger

DhineshAJ, 2017-05-21:
How to use hot words like "ok google" as trigger

ensonic, 2017-05-22:
To add another trigger, one needs to add a new trigger implementation here:
https://github.com/google/aiyprojects-raspbian/tree/master/src/triggers

Right now we don't have a hotword trigger in the example.

eduncan911, 2017-05-27:
looks like one of hte Google devs is already implementing it as a built-in trigger:

https://github.com/google/aiyprojects-raspbian/pull/64

I just pulled the branch down and have been testing it.  so far, it works great!

so I'd wait until it is merged.  then do a `git pull` to update your local copy.  make sure to run through the `scripts/install-deps.sh` again.  lots have changed since the original release.

sheridat, 2017-05-30:
Hi I have just added the following to #64 

I am a newcomer - so this question might not make a lot of sense but the sentence above "This requires installation of the wheel for the Google Assistant Library, which includes a closed-source hotword implementation" has me worried.

I associate wheel with updating python libraries via pip.

People are aware that there is a new version of the google assistant SDK is available BUT that it is for armv7 capable processors. People are also aware that it supports "hotword"

Many of us are running the AIY software on Pi Zero's and Pi W's which are armv6 based and therefore we cannot use that new version of the the assistant SDK.

My worry is that the change you are making here means that once it is merged - if I was to git pull this repository my installation will stop working because the google assistant will want an armv7 processor.

But as we say in england - I may be (entirely) barking up the wrong tree.

eduncan911, 2017-06-01:
PR #64 has been merged.  THis can be closed.

Update everything to latest, install the latest dependencies, and edit your voice-recorder.ini to use the "ok-google" trigger.

I also wrote up how to enable the OK Google hot words if anyone needs more step-by-step.

http://eduncan911.com/stem/embedded/aiy-google-assistant-upgrades-may-2017.html

closed 2017-07-11 by drigz