google/aiyprojects-raspbian #69: "pi" actions are not interrupted

eduncan911, 2017-05-27:
I recommend removing the "pi" prefix from the Actions, that was added under #30.  Or, pick a different prefix.  

How about "computer"? 
How about making it configurable?  We call ours "spee", so "spee power off" would work pretty well.

---

Per PR #30 , the new "power off" and "reboot" commands were prefixed with "pi":

"pi power off"
"pi reboot"

Unfortunately, our Voice Kit cannot interrupt "pi":

**5 year old daughter: "pi power off"**
`May 27 06:38:36 spee python3[550]: {'text': 'hey power off'}`

**me saying "pi power off"**
May 27 06:38:51 spee python3[550]: {'text': 'hi power off'}

**spouse saying "pi power off"**
May 27 06:39:52 spee python3[550]: {'text': 'say power off'}

Only after about a dozen times, was "Pi power off" detected by me saying it.  

Also, if you use commands like "hey power off", the assistant responds with *"Sorry, Power commands are not yet supported."*

^- this would indicate that the SDK may support power commands in the future itself through some API.  Something to keep an eye out for.

eduncan911, 2017-05-27:
@ktinkerer tagging ya here.

i may submit a PR if I get to it as well.

ensonic, 2017-05-30:
What about making it 'Raspberry power off' (and reboot likewise) or 'cube power off'?

eduncan911, 2017-05-30:
I think a configuration option, allowing the user to specify whatever prefix they like, would be best.

Or in my case, no prefix.  *"Hey Google, reboot"*  Because I use the hot words and it doesn't make sense to say, *"Hey Google, Pi reboot"*

Again I'm happy to do the work in a PR.  Just want some feedback/discussion here first.

---

I like the infrastructure that was done as part of the original PR to enable this.  It allows you to set up your own commands going forward.  So if you want to add things like:

```
Turn Left
Power Off TV
Where's the space station tonight?
```

If you had a custom prefix, and set to say Computer, you could do:

```
Computer Turn Left
Computer Power Off TV
Computer Where's the space station tonight?
```

Which would work pretty good, because *Computer* is a complex word to match.

It's just I have Hot Word activation enabled on ours, which wouldn't make sense:

```
Hey Google <pause> Computer Turn Left
Hey Google <pause> Computer Power Off TV
Hey Google <pause> Computer Where's the space station tonight?
```

Confuses my 5 year old when I tried linking our GoPiGo over the weekend. "But that doesn't make sense" she said.  lol

ktinkerer, 2017-05-30:
You can change the keywords in action.py to whatever you want:
 ```
actor.add_keyword(_('pi power off'), PowerCommand(say, 'shutdown'))
 actor.add_keyword(_('pi reboot'), PowerCommand(say, 'reboot'))
```

eduncan911, 2017-05-30:
the point of this Issue is to change it to something from that default, as it does not work very well.. with now 5 different people trying it (see OP).

closed 2017-05-31 by drigz

holten, 2017-06-03:
My take was "sudo power off" and "sudo reboot". =)

PR #73 (2017-05-31): Use "raspberry" instead of "pi" as power keyword
The speech recognition wasn't very good at recognizing "pi" but
"raspberry" seems to work better.

Fixes #69.
commit dd8ca35f (2017-05-31): Use "raspberry" instead of "pi" as power keyword (#73)

The speech recognition wasn't very good at recognizing "pi" but
"raspberry" seems to work better.

Fixes #69.