google/aiyprojects-raspbian #70: Feature request: change adding user's own commands in src/action.py to separate plugin py files

paulpoco, 2017-05-27:
Right now if an user wants to add their own commands to src/action.py they need to add their code to that file.

The problem is as corrections happen upstream it makes it harder to make sense of all the code in src/action.py

I propose to having a folder ~/voice-recognizer-raspi/src/plugins

Then each user can make their own commands or download commands made by others and place in ~/voice-recognizer-raspi/src/plugins

Then somehow ~/voice-recognizer-raspi/src/action.py needs to be changed to integrate the plugins on each start or restart. i.e. sudo systemctl restart voice-recognizer.service

pjbroad, 2017-05-29:
I've been working on something like to this on a folk here: https://github.com/pjbroad/aiyprojects-raspbian and would be happy for feedback and to do a pull request if suitable.

sheridat, 2017-05-29:
I agree. I pulled the upstream today and was surprised to find that commands concerning phillips hue bulbs had been added that needed supporting software packages to be installed.  Those commands are hardly generic.
Each user of AIY needs a mechanism to be able to define their own commands with import statements and each user should be responsible for installing the software packages that are needed to support their commands. 

I  use MPD for sound playback - and I have commands to use it's client MPC. I don't expect every other AIY user/owner to have MPD/MPC on their installation to satisfy my actions.

ensonic, 2017-05-30:
Hi,
This project was meant to be an example for how to integrate cloud apis into a raspberry pi based project. The main reason for the current structure was to keep things simple for newcommers -  just change one file. This is of course not scaleable and we alsodid not expect that many PRs to be honest. 

We will need to improve the code-structure to add a config file (see #45) and make actions/triggers plugins.

pjbroad, 2017-05-30:
@ensonic  My fork at https://github.com/pjbroad/aiyprojects-raspbian is an attempt to make user actions/trigger more like plug-ins.  It did not require much change to the existing code base either.  It might be worth taking a look.  I'd be happy take feedback and provide a pull request.

paulpoco, 2017-05-31:
To bad it the whole thing was run Debian package instead with upstream changes easily updated.

Make the system have a web interface that you could add your commands.  Similar to how Retropie has a web interface module for easily uploading ROMs.

I also guess the whole point of it was to use Python so us beginners could program it easier.

drigz, 2017-10-18:
Since the new direction of this project is to provide a support library (along with demos of how to use it) rather than a full-featured solution, I'm going to close this issue.

closed 2017-10-18 by drigz