google/aiyprojects-raspbian #72: How to update - some easy guidance for newcomers

sheridat, 2017-05-30:
Many of the people who have obtained the AIY are newcomers and will not be familiar with the wonderful world of Git. The article in MagPi regarding how to put the AIY kit together was very detailed and did a great job of holding peoples hands. 

The next hurdle for newcomers will be updating the software - Would it be possible to provide some guidance regrading how to:
1. Save any changes people have made to action.py
2. How to pull the updates

ensonic, 2017-05-30:
The steps would be:

    git add -m "<describe your change> src/action.py
    git pull --rebase

Now due to the nature of having everything in one file, one most likely get merge conflicts :/

eduncan911, 2017-05-30:
For the record, I have a blog post I am posting today to simplify how to update.  

Needs a bit more proof reading first... I'll come back to post it here.

eduncan911, 2017-05-30:
Published

http://eduncan911.com/stem/embedded/aiy-google-assistant-upgrades-may-2017.html

eduncan911, 2017-05-30:
@sheridat Published.

http://eduncan911.com/stem/embedded/aiy-google-assistant-upgrades-may-2017.html

sheridat, 2017-05-30:
**@eduncan911** Please tell your kids that your a gentleman and a scholar. Thank You so much for this - I've posted a link to it from the Raspberrypi.org forum for AIY.

eduncan911, 2017-05-30:
@sheridat np.  i've been meaning to this for a while, and this was an opportunity to sit down and kick it off.

i've tried registering at raspberrypi.org last week to help those with issues upgrading.  but alas, their registration system never sent me an email.

FAQs says "contact an Administrator for help during registration."

Help page says, "contact an Administrator for help."

Forum software developers says, "contact an Administrator for help."

Problem is, the only way to view the Team or admins - is with an account.

If you could drop them a line, give them my username "eduncan911", that would help.

sheridat, 2017-05-30:
Hi, sent them an email and copied you in. If you don't heart from them get
back to me and I will see if there is another route.

Terry

sheridat, 2017-05-31:
Hi, I followed the guidance in your blog.
At the step /scripts/install-deps.sh I got and error - the trace is below - in desperation I changed that line to sudo /scripts/install-deps.sh - it chucked out a message that it has to be run in pi - but it carried on and worked. 

Running virtualenv with interpreter /home/pi/voice-recognizer-raspi/env/bin/python3
Please use the *system* python to run this script
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/virtualenv.py", line 2328, in <module>
    main()
  File "/usr/local/lib/python3.4/dist-packages/virtualenv.py", line 713, in main
    symlink=options.symlink)
  File "/usr/local/lib/python3.4/dist-packages/virtualenv.py", line 925, in create_environment
    site_packages=site_packages, clear=clear, symlink=symlink))
  File "/home/pi/voice-recognizer-raspi/env/lib/python3.4/posixpath.py", line 357, in abspath
    if not isabs(path):
  File "/home/pi/voice-recognizer-raspi/env/lib/python3.4/posixpath.py", line 63, in isabs
    return s.startswith(sep)
AttributeError: 'NoneType' object has no attribute 'startswith'

eduncan911, 2017-05-31:
First, don't use `sudo`.  That may install the python dependencies as system packages, which is not what you want here because the `voice-recognizer` project runs under its own version of packages (the `env/bin` directory has its own version of the packages).  It didn't hurt anything.  Just run it again without the`sudo`.

I am not sure where you are seeing this error.  It's a python error, but the only python command in that shell script is:

`env/bin/pip install -r requirements.txt`

Try this:

    cd ~/voice-recognizer-raspi
    env/bin/pip install -r requirements.txt

And see if that completes.  If so, run `scripts/install-deps.sh` again.  If it still errors:

    cd ~/voice-recognizer-raspi/scripts
    install_deps.sh

Shouldn't have to though.

drigz, 2017-05-31:
@sheridat, @eduncan911: I got that error when I ran install-deps.sh from inside the "dev terminal". Currently it's designed to be run just once from a normal terminal.

Possible fix:
- adjust `install-deps.sh` to avoid recreating the virtualenv if it already exists
- on your page, move `install-deps.sh` before `activate`

sheridat, 2017-05-31:
@drigz  @eduncan911 
HI could you hold my hand a bit more - do you mean from inside the dev terminal rather than running this sequence
# UPDATE GOOGLE ASSISTANT SDK
cd ~/assistant-sdk-python
git checkout master
git pull origin master

# UPDATE VOICE RECOGNIZER (AKA: AIY PROJECTS)
cd ~/voice-recognizer-raspi
git checkout master
git pull origin master

# UPDATE DEPENDENCIES
cd ~/voice-recognizer-raspi 
source env/bin/activate
scripts/install-deps.sh

I should run the first two updates as is - and the 
# UPDATE DEPENDENCIES as follows
cd ~/voice-recognizer-raspi 
scripts/install-deps.sh
source env/bin/activate

Thank You both for your help

eduncan911, 2017-05-31:
@drigz  thanks for the input!  

Thinking back, yeah I did delete the `env` virtualenv a few times as I was upgrading and moving things around.  **I'll submit a PR tweak for that here** as the open PRs and other commits seem to be modifying this `install_deps.sh` file for the primary source of dependencies going forward (think system packages installs, as well as pip installs).  Makes sense to keep it all here, though I would think it belongs in the root directory.

I do everything via ssh remotely, no Dev Console.  

I didn't think activating the `virtualenv` would be an issue before running `install_deps.sh` (just good practice to keep someone always in a `virtualenv` - especially if they don't know Python and opening/closing terminals).  Within `install_deps.sh` it runs `sudo pip3` for system wide packages, and then purposely runs `env/bin/pip` for install in that location.  But you may be right that it is conflicting.  

**What if I modified the `install_deps.sh` to activate the virtualenv, instead of running `env/bin/pip`?**

That way, it continues with best practices and doesn't confuse newcomers trying to read what that script does (I actually didn't know you could run `env/bin/pip install` like that without a virtualenv until I saw that script in this repo - always thought you had to activate the virtualenv).

**What about a Makefile?**  

I could throw one together as well, that would also include `git pull` and such.

    make install
    make upgrade
    make run

Kind of thing.

---

@sheridat 

Sorry you having so much trouble as the upgrade process was a summation of my, troubles, getting things upgraded.  Yes, I think @drigz is correct that the directory already exists.  

This will work (close your terminal and start over).  

I see the "OK Google" hotwording has been merged in!  I just ran through these scripts and it worked:

_(yes, `git pull` the latest always - this is especially important for the `voice-recognizer-raspi` directory to do very very often)_

```
cd ~/assistant-sdk-python
git checkout master
git pull origin master

cd ~/voice-recognizer-raspi
git checkout master
git pull origin master

cd ~/voice-recognizer-raspi 
rm -rf env
scripts/install-deps.sh
```

And follow the rest of my blog post to enable hot wording and so on.

Also, I noticed the install script doesn't copy the new config files.  This is preferred, yes, so it doesn't override your existing config files.

But, if you want the latest config, you may want to:

    cp ~/.config/status-led.ini ~/.config/status-led.ini~
    cp config/status-led.ini.default ~/.config/status-led.ini
    cp ~/.config/voice-recognizer.ini ~/.config/voice-recognizer.ini~
    cp config/voice-recognizer.ini.default ~/.config/voice-recognizer.ini

And then open the config files and edit them again to your liking (e.g. `ok-google` option).

sheridat, 2017-05-31:
One more quicky please

I am about to do the sequence suggested - from "inside" a dev terminal

The bit that is

cd ~/voice-recognizer-raspi
rm -rf env
scripts/install-deps.sh

I am wondering whether - rm -rf env is deleteing the dev terminal that
I am working in? Should I come out of the dev terminal and use an
ordinary terminal session to do the above three commands? i am making
this up as I go along

eduncan911, 2017-05-31:
i don't think it would hurt anything.  `env/` is just the local packages installed for the python app to use.  It gets built from the original `scripts/install-deps.sh` execution you did a while ago.

I've deleted mine several times (just did it again before posting my last comment, to ensure it works).  and then rebuilt it with `scripts/install-deps.sh`.

I do everything from a non-Dev terminal (actually, remotely as I ssh in).  So if you run into any issues, yeah close the Dev terminal and run a normal one.

eduncan911, 2017-05-31:
Btw, I updated the blog post...

Added everything we talked about here.

Also added a new section on the latest `Config File` changes, and how to update them.

sheridat, 2017-05-31:
ur a star.

eduncan911, 2017-05-31:
nah, it's the guys actually doing the work here.  (they just merged the OK Google hot wording!  updated my post to reflect it's live!)

sheridat, 2017-05-31:
Just had my first game of trivia - fantastic!!!
Thank You everyone for all this help.

closed 2017-05-31 by sheridat