google/aiyprojects-raspbian #74: Ok Google - SDK upgrade - Here is the latest news - but there isn't any news

sheridat, 2017-06-02:
I've updated to a Pi 3 this morning and I've set the trigger to ok-google.  It works exceedingly well.
BUT
Normally I say good morning and it responds by saying hello, tells me the time and weather and what my commute is to get home and tells me to have a nice day

Now I have upgraded to the new version of the google assistant SDK i get the same response except after telling me about my commute and saying have a nice day - it says here is the latest news and then stops - it doesn't tell me any news and the journal shows it is waiting to be triggered.

That seems like a fault/bug to me

drigz, 2017-06-02:
Thanks very much for the report, @sheridat! I've reported this on to the Assistant SDK team - I'll update here when I know more.

mangelov95, 2017-06-14:
I have the same problem for google-assistant-sdk 0.3.0.

drigz, 2017-06-16:
The Assistant SDK team has fixed this, so the buggy "Here is the latest news" will be gone in the next release of [google-assistant-library](https://pypi.python.org/pypi/google-assistant-library) (presumably 0.0.3?)

Thanks again for the report!

closed 2017-06-16 by drigz