google/aiyprojects-raspbian #75: ValueError: list.remove(x): x not in list in audio.py

ywonglanecove, 2017-06-02:
Looks like endpointer_cb was fired more than it should have in main.py

Jun 03 08:55:20 raspberrypi python3[548]: return self._handle_response_stream(response_stream)
Jun 03 08:55:20 raspberrypi python3[548]: File "/home/pi/voice-recognizer-raspi/src/speech.py", line 206, in _handle_response_stream
Jun 03 08:55:20 raspberrypi python3[548]: self._end_audio_request()
Jun 03 08:55:20 raspberrypi python3[548]: File "/home/pi/voice-recognizer-raspi/src/speech.py", line 201, in _end_audio_request
Jun 03 08:55:20 raspberrypi python3[548]: self._endpointer_cb()
Jun 03 08:55:20 raspberrypi python3[548]: File "src/main.py", line 366, in endpointer_cb
Jun 03 08:55:20 raspberrypi python3[548]: self.recorder.del_processor(self.recognizer)
Jun 03 08:55:20 raspberrypi python3[548]: File "/home/pi/voice-recognizer-raspi/src/audio.py", line 77, in del_processor
Jun 03 08:55:20 raspberrypi python3[548]: self._processors.remove(processor)
Jun 03 08:55:20 raspberrypi python3[548]: ValueError: list.remove(x): x not in list

drigz, 2017-06-06:
Thanks for the report! I've seen this before, and added a check to avoid it, but I guess that doesn't always work.

What were you doing when this happened? Are you able to reproduce it reliably?

daver99999, 2017-06-24:
I am having the exact same issue. Tried a rebuild from scratch. No matter what I say it fails. Can anyone help?

daver99999, 2017-06-24:
When I push the button and talk the log sits at recognizing for like a minute and then bombs. here is my log:
Jun 24 05:50:36 raspberrypi systemd[1]: Starting voice recognizer...
Jun 24 05:50:36 raspberrypi systemd[1]: Started voice recognizer.
Jun 24 06:11:05 raspberrypi python3[534]: [2017-06-24 06:11:05,221] INFO:audio:started recording
Jun 24 06:11:05 raspberrypi python3[534]: [2017-06-24 06:11:05,257] INFO:main:ready...
Jun 24 06:43:22 raspberrypi python3[534]: [2017-06-24 06:43:22,697] INFO:main:listening...
Jun 24 06:43:22 raspberrypi python3[534]: [2017-06-24 06:43:22,700] INFO:main:recognizing...
Jun 24 06:45:54 raspberrypi python3[534]: [2017-06-24 06:45:54,373] INFO:speech:event_type: 1
Jun 24 06:45:54 raspberrypi python3[534]: [2017-06-24 06:45:54,385] INFO:main:thinking...
Jun 24 06:45:55 raspberrypi python3[534]: Exception in thread Thread-2:
Jun 24 06:45:55 raspberrypi python3[534]: Traceback (most recent call last):
Jun 24 06:45:55 raspberrypi python3[534]: File "/usr/lib/python3.4/threading.py", line 920, in _bootstrap_inner
Jun 24 06:45:55 raspberrypi python3[534]: self.run()
Jun 24 06:45:55 raspberrypi python3[534]: File "/usr/lib/python3.4/threading.py", line 868, in run
Jun 24 06:45:55 raspberrypi python3[534]: self._target(*self._args, **self._kwargs)
Jun 24 06:45:55 raspberrypi python3[534]: File "src/main.py", line 377, in _recognize
Jun 24 06:45:55 raspberrypi python3[534]: self._handle_result(self.recognizer.do_request())
Jun 24 06:45:55 raspberrypi python3[534]: File "/home/pi/voice-recognizer-raspi/src/speech.py", line 259, in do_request
Jun 24 06:45:55 raspberrypi python3[534]: return self._handle_response_stream(response_stream)
Jun 24 06:45:55 raspberrypi python3[534]: File "/home/pi/voice-recognizer-raspi/src/speech.py", line 206, in _handle_response_stream
Jun 24 06:45:55 raspberrypi python3[534]: self._end_audio_request()
Jun 24 06:45:55 raspberrypi python3[534]: File "/home/pi/voice-recognizer-raspi/src/speech.py", line 201, in _end_audio_request
Jun 24 06:45:55 raspberrypi python3[534]: self._endpointer_cb()
Jun 24 06:45:55 raspberrypi python3[534]: File "src/main.py", line 366, in endpointer_cb
Jun 24 06:45:55 raspberrypi python3[534]: self.recorder.del_processor(self.recognizer)
Jun 24 06:45:55 raspberrypi python3[534]: File "/home/pi/voice-recognizer-raspi/src/audio.py", line 77, in del_processor
Jun 24 06:45:55 raspberrypi python3[534]: self._processors.remove(processor)
Jun 24 06:45:55 raspberrypi python3[534]: ValueError: list.remove(x): x not in list

ywonglanecove, 2017-06-25:
I got these when I had regular conversation using clap trigger I haven't
seen it again because I addedcheck before this line but that seems not a
proper way to fix this

drigz, 2017-06-27:
@ywonglanecove and @daver99999, thanks for the report and for the complete log.

I've looked into this but can't reproduce. The correct order of events is:

1. Clap.
1. SyncMicRecognizer.recognize() get called.
1. recognize() calls recorder.add_processor().
1. recognize() sets recognizer_event
1. The background thread, _recognize() handles recognizer_event and takes over.
1. _recognize() calls recognizer.do_request(), which calls recognizer._handle_response_stream().
1. The server detects that the user has stopped speaking, and returns the `END_OF_UTTERANCE` event.
1. _handle_response_stream() gets this response and calls _stop_sending_audio(resp).
1. _stop_sending_audio() prints `event_type: 1` then returns True.
1. _handle_response_stream() calls _end_audio_request(), which calls endpointer_cb().
1. endpointer_cb() calls del_processor(recognizer).
1. endpointer_cb() calls status('thinking') which prints `thinking...`.
1. The server sends further responses with the transcript and audio data.

The problem appears to be that del_processor() is called more than once, because `thinking...` appears in davey's log, and it is printed after the first call to `del_processor()`. It is possible that the server returns an error after sending `END_OF_UTTERANCE`, which would lead to `_end_audio_request` being called twice. However, I can't think what would cause this this.

If that's the problem the right fix is to add state to `GenericSpeechRequest` so it only calls endpointer_cb() once per request.

@ywonglanecove, I wouldn't mind also adding a check to turn the crash into a warning. Would you like to make a PR with your check? If not I can do it.

ywonglanecove, 2017-06-28:
>    2. SyncMicRecognizer.recognize() get called.
>    3. recognize() calls recorder.add_processor().
>    4. recognize() sets recognizer_event
>    5. The background thread, _recognize() handles recognizer_event and
>    6. _recognize() calls recognizer.do_request(), which calls
>    7. The server detects that the user has stopped speaking, and returns
>    8. _handle_response_stream() gets this response and calls
>    9. _stop_sending_audio() prints event_type: 1 then returns True.
>    10. _handle_response_stream() calls _end_audio_request(), which calls
>    11. endpointer_cb() calls del_processor(recognizer).
>    12. endpointer_cb() calls status('thinking') which prints thinking....
>    13. The server sends further responses with the transcript and audio

I will create a pr in the weekend

ywonglanecove, 2017-07-15:
Closed as PR merged

closed 2017-07-15 by ywonglanecove