google/aiyprojects-raspbian #79: MagPI AIY - Ok Google - New SDK - new functionality not available if alternative triggers used

sheridat, 2017-06-06:
Hi, So now we have successfully managed to update our cardboard friend - thanks to the wonderful guidance yourselves provided - we have what seems to be another problem/issue. 

In eduncan911's blog it was mentioned that the timer functionality was included in the new assistant SDK .
So we have found that 
IF you setup your trigger to be ok-google - and ask the assistant to set a timer - the assistant will indeed set a timer and after the elapsed time it will tinkle away until you say ok-google stop.
HOWEVER 
IF you leave that trigger at the default - which of course is our beloved pi GPIO - the timer functionality does not work - oh nooooooooooooooo
I have an awful feeling THAT (even if you have an PI 2/3 and hence you can use the new SDK) you are only using that new SDK (and its new functionality such as the timer) if you setup to use the hotword trigger. 
I have an even worse feeling that whilst the new SDK knows about hotword triggering - it is clueless about GPIO.
Some people are going to want to be able to trigger via GPIO and be able to set a timer and whatever else the google-assistant is able to do down the road. Given the new SDK is only available for the PI it seems inconceivable it isn't usable with GPIO which is fundamental to the PI's being.

I've mad some assumptions and maybe be barking up the wrong tree again. Look forward to hearing from you

drigz, 2017-06-06:
Thanks for the feedback, @sheridat!

This is intentional - we were trying to keep things a bit simpler for the user by hiding the differences between the Google Assistant API (used for clap/GPIO trigger) and the Google Assistant Library (used for OK Google). However, it sounds like the missing functionality with the Assistant API means this isn't actually any simpler.

We're working on a new design at the moment that will be easier for people to build into their own applications, so this is great feedback to have. 

Anyone reading this who wants to use timers etc with GPIO triggering - please +1 @sheridat's post so we know you're interested.

sheridat, 2017-06-06:
Thank You for the reply - this is my +1. People have come up with a few ways to trigger the assistant such as dash buttons and scripts. I trigger it by script (that sets the gpio) and in my instance I run that script using an IR remote (using LIRC/irexec)
We really want any new functionality - plus flexible triggering

borpin, 2017-06-06:
+1 (even though I'm still on a Zero so cannot use it as yet.)  I really do not understand the differences between the library and the API and what the separation is trying to achieve.

czarnikjak, 2017-06-06:
+1 sheridat. I second that...there should be the same functionality regardless of how you trigger. Otherwise it gets very messy. Thanks

janpIFMS, 2017-06-06:
@sheridat I've also witnessed this.
It's almost like two different responses are active but the pi only plays one.

JonesChi, 2017-06-16:
+1 for this feature. I'm currently using the customized trigger with snowboy, and the timers/alarms does not work.

arman-k, 2017-07-15:
Any update on this?

nskrotzki, 2017-08-29:
Has there been any fixes to this?

drigz, 2017-08-30:
Nope, no update, I'm afraid. #112 contains an example of GPIO trigger with the assistant library, but you'd have to add your custom commands yourself, as the example doesn't use actions.py.

aycgit, 2017-09-15:
@JonesChi can you post a gist of that? I am trying to setup mine but i can't get it to work.

drigz, 2017-10-18:
The [assistant_library_with_button_demo.py](https://github.com/google/aiyprojects-raspbian/blob/b47767e2a5d7c2495900a46aa563846a2f631d5e/src/assistant_library_with_button_demo.py) will allow timers and alarms with a GPIO trigger.

closed 2017-10-18 by drigz

Shulyaka, 2020-06-22:
Hi! It looks like the home assistant library is deprecated now. How do I now set a timer with python?

Shulyaka, 2020-06-22:
There should probably be a timer trait