google/aiyprojects-raspbian #80: Multiple triggers

sheridat, 2017-06-07:
Hi, One of the principal use cases of the AIY kit is to play music - including music stashed in google play servers. Some people have even soldered on another DAC/AMP so they can listen in stereo. 

However if you use hotword triggering - once your playback is above a certain volume level you end up being unable to **reliably** trigger the assistant to ask for the volume to be lowered or move to the next track etc (various people including me have implemented actions to control our favourite music players). The microphone/software can't hear the ok google trigger phrase.  

I noticed in http://eduncan911.com/stem/embedded/aiy-google-assistant-upgrades-may-2017.html made a comment that currently the code only supports one trigger.

Any ideas how the catch 22 can be resolved - if it can be resolved? I am guessing that being able to trigger by another means - such as GPIO that lowers the volume/mutes the volume is an obvious beginning.   

Oh I am aware of  - don't play your music so loud :-) Thank You in advance

eduncan911, 2017-06-08:
I agree there should be multiple triggers available.

But note: if the Voice Kit/Google Home device can't hear your "Ok Google" - which has gone through **very rigorous** pattern matching across many dialects and age groups, it most likely won't understand what you say next after triggering it with such loud volumes of music.

Just saying.  👍  

One thing to play with is the Gain controls of the mic input.  As noted in #47 , the gain control with the AIY Projects SD Card image is set too high and distorts (clips) the input if too close/too loud.  

We (my family) have seen this a lot.  

I can be in the next room and speak with just a slight kick in my voice, and the Voice Kit will hear my "Ok Google" and pick up what I am asking it.  Again, from another room!  Very very sensitive!

On the other hand, with my 5 year daughter standing right next to it and saying, "Ok Googles" - it never hears her.  I ask her to speak much quieter (knowing the gain is really high), and it hears her.  

Same goes with me.  If I am speaking slightly higher, say over a boiling teapot, the Voice Kit never hears me.  If tone it down, to a "it should never be able to hear me at this volume, in the kitchen, on the other side of a wall, with a boiling teapot - it hears me!!!!!

I've tweaked the gain to +10 as noted in #47 and it works, with her speaking pretty loud next to it and I can speak up and it hears me.  But I now kind of miss the sensitivity I had before.

sheridat, 2017-06-08:
Thanks for such a detailed reply. I will have a good play with the gain controls and read  #47. 

If we were able to trigger the assistant by GPIO as well as by hotword we can solve most of the issues. 

Some of us are already using scripts to trigger the assistant by manipulating GPIO pin 23. A good example is using a dash button which when pressed fires up the assistant. See post "https://www.raspberrypi.org/forums/viewtopic.php?f=114&t=184283". 

In my case when I was using the default GPIO trigger and playing music,  I used to get the assistant to fire up by running a python script that muted/unmuted my music player using a toggle. and then set GPIO 23 as per the post above. I would ask the assistant whatever and the when ready ask her to play music again. Rather than a dash button - I triggered my script by IR remote but I've also used ssh to run it and well the possibilities open up to solve things like this if we can use an alternative trigger when we have set the default to ok-google. 

As an aside - I've just been to my local electronic shop and bought everything to make an "ir blaster" so by teatime I should be able to say "ok google TV volume up" and so on.  I love this thing - it has really captured my imagination.

sckam, 2017-08-07:
Dear all,

I have my AIY Project. I want to turn ON and OFF two digital outputs at the same time.
Here my code:

**chan_list = (17,27) # the GPIO 17 and 27
actor.add_keyword('on everything', GpioWrite(17 & 27, True))  # all on
actor.add_keyword('off everything', GpioWrite(17 & 27 False)) # all off

actor.add_keyword('turn on all device',  GPIO.output(chan_list, gpio.high) # all on
actor.add_keyword('turn off all device', GPIO.output(chan_list, gpio.low)  # all off**

I have try the above method, it only turn on or off GPIO 17, cannot turn on both 17 and 27 at the same time.

Kindly advise me.
Thank you.

ensonic, 2017-08-07:
@sckam please use a new ticket for new issues. In your cases GpioWrite(17 & 27, True) does not do what you think it would. GpioWrite can only affect a single pin. You are invoking a bitwise operator: https://wiki.python.org/moin/BitwiseOperators. Conicidentially 17 & 27 is still 17 (all the bits set in 17 are also set in 27). The easiest fix is to duplicated the add_keyword lines for each pin.

sckam, 2017-08-08:
Hi ensonic,
Thanks for the advice, I will create new ticket, next time.

I have try like this early, as what you advise:

**actor.add_keyword('on everything', GpioWrite(17, True))
actor.add_keyword('on everything', GpioWrite(27, True))
actor.add_keyword('off everything', GpioWrite(17, False))
actor.add_keyword('off everything', GpioWrite(27, False))**

When I say "on everything" or "off everything" only GPIO 17 is responding, GPIO 27 have no response.
Unless I say on or off individually, for example on GPIO 27, then GPIO 27 have the response.
It looks like when say the common term: "on everything" or "off everything", the GPIO 17 execute priority, compare to GPIO27. When I modify the code sequence slightly:

actor.add_keyword('on everything', GpioWrite(27, True))
actor.add_keyword('on everything', GpioWrite(17, True))
actor.add_keyword('off everything', GpioWrite(27, False))
actor.add_keyword('off everything', GpioWrite(17, False))

Then when I say "on everything" or "off everything" only GPIO 27 is responding, GPIO 17 have no response.

So confirm it is due to the sequence priority, when I say "on everything" or "off everything".

Any ideal how to overcome this ? 
Possible have any work around method ?

eduncan911, 2017-08-08:
@sckam please create a new Issue.  this issue is for something totally different.

i have suggestions, but they don't belong here.

drigz, 2017-10-18:
There's an example for this in the new [assistant_library_with_button_demo.py](https://github.com/google/aiyprojects-raspbian/blob/b47767e2a5d7c2495900a46aa563846a2f631d5e/src/assistant_library_with_button_demo.py).

closed 2017-10-18 by drigz