google/aiyprojects-raspbian #81: How to set up French Language for Google Assistant AIYProject ?

anneromane, 2017-06-09:
hello,

How to set up French Language for Google Assistant AIYProject ?

Will we have the voice in French?

Thks 

Regards

Anne Romane

alfongj, 2017-06-09:
Hi Anne Romane. 

French is currently not supported on the Google Assistant API, but it will come in the future. We don't have a release date yet.

If you only want to do voice commands, you can do that with the Cloud Speech API (but it won't have Assistant responses)

drigz, 2017-06-27:
Closing, as this is an issue in the backend.

As @alfongj says, if you use the Cloud Speech API ([instructions](https://aiyprojects.withgoogle.com/voice/#makers-guide-3-1--change-to-the-cloud-speech-api)) and change the language in `~/.config/voice-recognizer.ini` you can create your own French voice commands. If you'd like, you could create a file `fr.po` alongside [de.po](https://github.com/google/aiyprojects-raspbian/blob/master/po/de.po) ([instructions](https://github.com/google/aiyprojects-raspbian/blob/master/HACKING.md#i18n)). Then, you could translate the existing commands to French, and send us a pull request so other users get this.

closed 2017-06-27 by drigz

chris777c, 2017-12-17:
Hello,

With new release of project , voici-recognizer.ini don't exist ? How do set up french language on last version AIY project ?

sheridat, 2017-12-18:
@chris777c in your code use this setting
 .i18n.set_language_code('fr-FR') 
alternatively you can set it each time you use aiy.audio.say
aiy.audio.say("I ran the script", lang='fr-FR')

yonisolo, 2018-01-17:
@sheridat that's exactly what i've been looking for since weeks

sheridat, 2018-01-17:
@yonisolo  Hi things have moved on a bit. The latest version of the assistant can speak more languages.

If your on the voicekit branch - i believe it has been updated to speak other languages

In time the aiyprojects branch will be updated 

I don't need the capability - see https://github.com/google/aiyprojects-raspbian/pull/235#issuecomment-357453352 which has some more info.

chris777c, 2018-03-01:
Hello,

I reopen subject because the 235 post speak "dirty solution". Does it exist a good solution for setup french assitant in AIY projet ?

drigz, 2018-03-02:
@chris777c Give it a try with the [latest image](https://dl.google.com/dl/aiyprojects/aiyprojects-latest.img.xz). Using the Assistant Library, you can change the language with the Google Assistant app. Using the Assistant Service (assistant_grpc_demo.py) you can change the language with the app or with `aiy.i18n.set_language_code('fr-FR')`.

Let us know how that works out for you!

chris777c, 2018-03-02:
Hello drigz, how to install "latest image" on AIY already install ?

Jaxom99, 2018-03-13:
Hi @drigz and @sheridat : I reckon you managed to use the Google Assistant in French (speech and recognition). Is this correct ? I searched everywhere to get it to work (RPi3 and Google AIY Kit, no android phone), but I feel like I'm missing something...
Can you try and make an updated step-by-step, with the right branch to use ?

gabrielklein, 2018-03-15:
(Based on the feedbacks given in this thread.)

cd /home/pi/AIY-projects-python
cp src/examples/voice/assistant_grpc_demo.py src/examples/voice/assistant_grpc_demo_fr.py

Edit 
src/examples/voice/assistant_grpc_demo_fr.py

Between
>    status_ui.status('starting')
>    assistant = aiy.assistant.grpc.get_assistant()

Add
>     status_ui.status('starting')
>     aiy.i18n.set_language_code('fr-FR')
>     assistant = aiy.assistant.grpc.get_assistant()

Jaxom99, 2018-03-15:
Thanks @gabrielklein ! I tried this before, but maybe too late in the script so it kept speaking English. My workaround for now has been to re-install the whole project using the provided image.
I think the "Hacking.md" way is really obsolete, so I was missing some steps or some new versions of something...
I will try again later on, and let you know if it works.

yonisolo, 2018-03-19:
Hi there :)
I tried this weekend, and nope it doesn't work, it seems that the way the set_language is implemented is not very clear...
However, i saw some quest challenge to test with grpc but i didn't have time to...
please see: https://github.com/googlesamples/assistant-sdk-python/issues/146 for details about it!
I'll give it a try later if nobody tried before.
Thanks for keeping this thread open everybody!!!

fabbbbb, 2019-01-15:
> @chris777c  Using the Assistant Library, you can change the language with the Google Assistant app. 

How do you manage to do this?
I want to modify the language for[assistant_library_with_button_demo.py](https://github.com/google/aiyprojects-raspbian/blob/voicekit/src/assistant_library_with_button_demo.py)

yonisolo, 2019-01-15:
I sent an email to the aiy project support and they answered me:

> Thank you for your interest into AIY projects.

> Please refer : https://developers.google.com/assistant/sdk/reference/library/languages to know what all 
> languages are supported by Assistant grpc demo. 

> You can read about python APIs to change the laguage at : https://aiyprojects.withgoogle.com/voice /#makers-guide-python-api-reference

yonisolo, 2019-01-16:
Hi @dmitriykovalev, thank you for your reply, that's really great!!
Some time ago I tried with this code parameters, and it didn't worked at all. 
The best i managed to get is speak in french and answered in english, after numbers of tries and without knowing exactly what I did to get it...
Did something change about it in the last Image? 
This language problem is so dark where it should be easy...

I tried with this piece of code:
> aiy.i18n.set_language_code('fr-FR')
> assistant = aiy.assistant.grpc.get_assistant()

yonisolo, 2019-01-17:
@dmitriykovalev that's a great news!!! I'll give it a try and post my result

fabbbbb, 2019-01-17:
For library, you have to connect a Google Assistant App with a smartphone (with the same account as on the device) to your device. And then modify the language with this App (tested and works on assistant_library_with_button_demo.py).  
For other scripts, just add `--language` (i.e. fr-FR) to the command line when you launch your script.