google/aiyprojects-raspbian #86: change default sound volume

c8001221, 2017-06-22:
Each time when I start the service, I found that the sound volume setting has been reset again and a bit loud. Anyway I can set a default sound volume except by voice command each time?

thomas-vl, 2017-06-22:
You can set the default pi volume with this command:

amixer  sset PCM,0 70%

Now every time you reboot it should be set to 70%.
Please note if you don't shut down or reboot properly this setting won't be saved. Try and avoid pulling the power plug.

drigz, 2017-06-27:
@c8001221, if you reboot safely (eg with the voice command `raspberry reboot` or `raspberry power off`, or `sudo reboot` on the command line) does it still forget your setting?

For me, if I change the volume and reboot safely, the volume setting isn't changed.

drigz, 2017-07-11:
Can't reproduce - closing.

closed 2017-07-11 by drigz