google/aiyprojects-raspbian #88: Memory Leak error

eduncan911, 2017-06-25:
Occassionally, once a week or more frequenty, when we say, *Hey Google*, she responds instantly with:

> Sorry, something went wrong.  Please try again in a few seconds.

And many other variants similar to this to "retry" in a few seconds.

Reviewing the journal log, I found this:

```
Jun 25 09:22:43 spee python3[13710]: [2017-06-25 09:22:43,596] INFO:root:ON_CONVERSATION_TURN_STARTED
Jun 25 09:22:43 spee python3[13710]: [2017-06-25 09:22:43,597] INFO:main:listening...
Jun 25 09:22:43 spee python3[13710]: E0625 09:22:43.971572259   15218 network_status_tracker.c:48] Memory leaked as all network endpoints were not shut down
Jun 25 09:22:43 spee python3[13710]: [13710:15218:ERROR:speech_processing_task.cc(646)] Detected communication error: Downstream closed prematurely.
Jun 25 09:22:43 spee python3[13710]: [13710:15218:ERROR:assistant_output_handler.cc(589)] Failed with a general speech recognition error
Jun 25 09:22:43 spee python3[13710]: [2017-06-25 09:22:43,973] INFO:root:ON_ASSISTANT_ERROR:
Jun 25 09:22:43 spee python3[13710]: {'is_fatal': False}
Jun 25 09:22:43 spee python3[13710]: [2017-06-25 09:22:43,974] INFO:root:ON_RESPONDING_STARTED:
Jun 25 09:22:43 spee python3[13710]: {'is_error_response': True}
Jun 25 09:22:43 spee kernel: voicehat-codec voicehat-codec: Enabling audio amp...
Jun 25 09:22:47 spee python3[13710]: [2017-06-25 09:22:47,821] INFO:root:ON_RESPONDING_FINISHED
Jun 25 09:22:47 spee python3[13710]: [2017-06-25 09:22:47,822] INFO:root:ON_CONVERSATION_TURN_FINISHED:
Jun 25 09:22:47 spee python3[13710]: {'with_follow_on_turn': False}
Jun 25 09:22:47 spee kernel: voicehat-codec voicehat-codec: Disabling audio amp...
Jun 25 09:22:47 spee python3[13710]: [2017-06-25 09:22:47,823] INFO:main:ready...
```

Do note that the speech that comes back is the Google-streamed or "Real" Assistant SDK voice - not the cheap-robot-sounding local commands version of the voice.  So it is reaching the SDK.

drigz, 2017-06-26:
If you're using hotwording, there's not a lot of control we have - I think you're right about it being an issue in the SDK. Let's see what they say there.

> Do note that the speech that comes back is the Google-streamed or "Real" Assistant SDK voice

It is possible that the Assistant library returns a prerecorded clip of assistant speech when it can't reach the server - having a good voice doesn't necessarily mean that it's not a problem with the network.

Does it work when you retry?

eduncan911, 2017-06-26:
> Does it work when you retry?

Nope.  It is permanently broken until you restart the services.  It remains broken for hours or even days until you restart.

It isn't a network issue.  Well, maybe it was the first time perhaps - I don't know as I wasn't recording pings (am now).  But it never corrects itself.

eduncan911, 2017-08-08:
@drigz I am indeed leaning towards that it is network related as the voice kit will be working fine the previous night, I come downstairs in the morning and say, "Hey Google" and she pauses for about 30 seconds, glowing that button, and coming back and speaking the error message.

Any retry after that instantly comes back with the error reply.

Wifi networking using the Pi 3's built in (yeah, not that strong).  Out of 3 or 4 weeks, it is solid.  Just rarely happens.  

I've setup pings but usually kill it or reboot the machines doing the pings because of the weeks-between-issue.  Though, I have noticed there are dropped pings of about 0.1% over several days.  

So perhaps it is when these dropped pings are happening, and the timing of me trying to use Google Assistant, that it fails.  Rare, but it does happen.

drigz, 2017-08-08:
@eduncan911 Thanks for the extra info - could you cross-post it to the other issue? I'm going to close this one as I'm 99% sure it's an issue in the underlying google-assistant-library, probably caused by the network somehow (I've never seen it here).

closed 2017-08-08 by drigz