google/aiyprojects-raspbian #89: spurious triggering - hotword

sheridat, 2017-06-26:
The AIY box when setup for hotword triggering is reacting to spurious phrases.

I now have my AIY setup to trigger using hotword.  Like most people I don't spend my day watching it's pulsing LED but I have been assuming that it is only triggered at my behest.

However today I switched on the trigger sound facility - using a wav file already on the pi. So my config looks like this
# Select the trigger: gpio (default), clap, ok-google.
 trigger = ok-google
# Select the trigger sound:
 trigger-sound = /usr/share/scratch/Media/Sounds/Electronic/Fairydust.wav

A couple of times today I have just been sitting watching tennis on the TV and the AIY has suddenly triggeredand the wav file has been played.  The latest example was this example from the journal - it was just sitting there and I didn't trigger it by using either Ok Google or Hey Google
Jun 26 **19:59:24** raspberrypi python3[522]: [2017-06-26 19:59:24,082] INFO:main:ready...
Jun 26 20:14:38 raspberrypi python3[522]: [2017-06-26 20:14:38,053] INFO:root:ON_CONVERSATION_TURN_STARTED
Jun 26 20:14:38 raspberrypi python3[522]: [2017-06-26 20:14:38,054] INFO:main:listening...
Jun 26 20:14:48 raspberrypi python3[522]: [2017-06-26 20:14:48,069] INFO:root:ON_END_OF_UTTERANCE
Jun 26 20:14:48 raspberrypi python3[522]: [2017-06-26 20:14:48,071] INFO:main:thinking...
Jun 26 20:14:48 raspberrypi python3[522]: [2017-06-26 20:14:48,333] INFO:root:ON_RECOGNIZING_SPEECH_FINISHED:
Jun 26 20:14:48 raspberrypi python3[522]: {'text': "that's pretty old and hasn't for love and it difficult for Everything Jewish results"}
Jun 26 20:14:49 raspberrypi python3[522]: [2017-06-26 20:14:49,765] INFO:root:ON_CONVERSATION_TURN_TIMEOUT
Jun 26 20:14:49 raspberrypi python3[522]: [2017-06-26 20:14:49,766] INFO:root:ON_CONVERSATION_TURN_FINISHED:
Jun 26 20:14:49 raspberrypi python3[522]: {'with_follow_on_turn': False}
Jun 26 20:14:49 raspberrypi python3[522]: [2017-06-26 20:14:49,768] INFO:main:ready...

The context of the tennis match certainly didn't involve anything that could be construed as either Ok Google or Hey Google. I can understand that the phrase **that's pretty old and hasn't for love and it difficult for Everything Jewish results** does contain a couple of words that are tennis related.

drigz, 2017-06-27:
There's nothing we can do about this, I'm afraid. False triggering is pretty much inevitable with current hotwording implementations. Some others (but not the Google Assistant Library) let you adjust the sensitivity, but then you'll miss more real triggers.

closed 2017-06-27 by drigz

sheridat, 2017-06-27:
And here was I thinking I had found "the ghost in the machine"