google/aiyprojects-raspbian #94: Adding Google AIY to DietPi

R1CH-T, 2017-07-10:


R1CH-T, 2017-07-10:
```
root@AIY:/mnt/dietpi_userdata/voice-recognizer-raspi# systemctl status voice-recognizer.service
● voice-recognizer.service - voice recognizer
   Loaded: loaded (/lib/systemd/system/voice-recognizer.service; enabled)
   Active: failed (Result: start-limit) since Mon 2017-07-10 01:24:53 UTC; 15min ago
  Process: 583 ExecStart=/mnt/dietpi_userdata/voice-recognizer-raspi/env/bin/python3 -u src/main.py (code=exited, status=217/USER)
 Main PID: 583 (code=exited, status=217/USER)
   CGroup: /system.slice/voice-recognizer.service

root@AIY:/mnt/dietpi_userdata/voice-recognizer-raspi# cat /etc/asound.conf
options snd_rpi_googlemihat_soundcard index=0

pcm.softvol {
    type softvol
    slave.pcm dmix
    control {
        name Master
        card 0
    }
}

pcm.micboost {
    type route
    slave.pcm dsnoop
    ttable {
        0.0 30.0
        1.1 30.0
    }
}

pcm.!default {
    type asym
    playback.pcm "plug:softvol"
    capture.pcm "plug:micboost"
}

ctl.!default {
    type hw
    card 0
}

root@AIY:/mnt/dietpi_userdata/voice-recognizer-raspi# aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
softvol
micboost
default
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
dmix:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample mixing device
dmix:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample mixing device
dsnoop:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample snooping device
hw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct hardware device without any conversions
plughw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Hardware device with all software conversions

root@AIY:/mnt/dietpi_userdata/voice-recognizer-raspi# arecord -l
**** List of CAPTURE Hardware Devices ****
root@AIY:/mnt/dietpi_userdata/voice-recognizer-raspi# arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
softvol
micboost
default
root@AIY:/mnt/dietpi_userdata/voice-recognizer-raspi# aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
softvol
micboost
default
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
dmix:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample mixing device
dmix:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample mixing device
dsnoop:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample snooping device
hw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct hardware device without any conversions
plughw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Hardware device with all software conversions
```

ensonic, 2017-07-10:
Could you run the checkpoint scripts:
https://github.com/google/aiyprojects-raspbian/tree/master/checkpoints

R1CH-T, 2017-07-10:
Thank you for the reply, ensonic.

I can't look at it again, right now, but i actually did manually run all the checkpoint scripts and if I recall correctly, they all failed. (I was up very late last night fiddling around with this thing.)

I'll try the `lsmod` ASAP, too.

I've been given a couple of pointers [over at my other issue page](https://github.com/Fourdee/DietPi/issues/1065), so I'll try those suggestions, too.

R1CH-T, 2017-07-10:
I think we've identified the problem: It's to do with file paths, mainly.

Dietpi uses the directory /DietPi/ to hold a ramdisk and copy of config.txt. The changes need to be applied there.

Also, there are a few hard-coded paths in the code, which will need rectifying at your end, I think:

https://github.com/Fourdee/DietPi/issues/1065#issuecomment-314157213

Thanks,
Rich.

R1CH-T, 2017-07-10:
Just to let you know; I've got it working, with the exception of the system services, because they all contain hard-coded links to `/home/pi`. This will need sorting out to ensure better compatibility with other distros, but, yeah; works like a charm!

Thanks for your input.

drigz, 2017-07-11:
`install-services.sh` uses `sed` to change the path in the service files to the correct value, so it _should_ be portable. How did it fail in your case?

`install-alsa-config.sh` could be generic - I've filed #95 to track this, thanks for finding it!

R1CH-T, 2017-07-11:
Thanks so much for replying. :+1: 

> install-services.sh uses sed to change the path in the service files to the correct value, so it should be portable. How did it fail in your case?

I'm not a coder _at all_ and even my rudimentary script-kiddie skills are not up to the task of keeping up with you guys, but I think that the fact that the sed command in question starts  `sed "s:/home/pi/voice-recognizer-raspi`... might be a problem:
...
for service in systemd/*.service; do
  sed "s:/home/pi/voice-recognizer-raspi:${repo_path}:g" "$service" \
    > "/lib/systemd/system/$(basename "$service")"
done
...
Should it use `$HOME`? Or maybe some variation of the line from the same script, which sets the base directory:

`cd "$(dirname "${BASH_SOURCE[0]}")/.."`

...and go up one level from that; but apply for _all of the system services_.

I'm not sure if this is viable and I'm not sure that I even know what I'm saying, myself; but rectifying this could potentially prevent (some) future issues of this kind.

In the case of DietPi, everything is run under Root, so all stuff directed to `/home/user/` ends up in `/root/` and all stuff directed to `/home/pi/` fails. Hence the problem. This was the author's decision and I'm pretty sure that this has been debated, but I'm not qualified to weigh in. The point is that if anything is hard-coded to point to, or to look in /home/pi/ (or any other arbitrary directory); chaos ensues. If `/voice-recognizer-raspi/` was dropped into `/home/richard/`, we would be dealing with the same sort of problem.

Please forgive me if I am rattling on, but I need to educate myself and I hope that I can, in some small way, help both of the projects that I dragged into this issue. It's just that it's so easy and so much of a time-saver, to use Dietpi to set up and connect applications and services, that I find it hard to beat. Also, I have had so much more fun playing around with the AIO kit than I ever did with using the Google Assistant on a phone.

drigz, 2017-07-11:
> I think that the fact that the sed command in question starts sed "s:/home/pi/voice-recognizer-raspi... might be a problem:

The reason for this is that the files themselves contain "/home/pi/..." in their current state. This command should replace "/home/pi/..." with the correct path "${repo_path}" as it installs the scripts. You can try running the script and looking at `/lib/systemd/system/voice-recognizer.service` to see if it has the right paths.

However, you will need to manually edit the files `voice-recognizer.service`, `status-led.service` and `status-monitor.service` in `/lib/systemd/system/` to remove the line `User=pi`. (if this works for you, that could be made automatic in install-services.sh)

You may also have other issues running as root - I seem to remember someone trying this before and having some problems before getting it working, but can't find the thread. Let us know how it goes!

> Please forgive me if I am rattling on

Not at all - we're happy to have the feedback!

R1CH-T, 2017-07-12:
Well, I got it all working now. I just had to add a few sed commands to the installer code to change the file paths and remove `user=pi` from the relevant services. If you are interested, look here:

https://github.com/Fourdee/DietPi/issues/1065#issuecomment-314837634

The only issues I have now are a lack of Power Off and Reboot commands and the silent alarm, mentioned in another issue, here.

Thanks,
Rich.

drigz, 2017-10-18:
I think #156 resolved the issue with the new voicekit branch, so hopefully everything's working now.

closed 2017-10-18 by drigz