google/aiyprojects-raspbian #96: Add script to restart all of aiy services

LegoChicken, 2017-07-12:
Add script to restart all of aiy services. Useful to restart after modifying the source code. Would be quicker than restarting the Raspberry Pi.

sheridat, 2017-07-13:
Don't you just need to type 'sudo systemctl restart voice-recognizer' to restart? If you do type that it stops the service and then starts it again - total elapsed time is 8 seconds on my aiy box.  That's how I pick up changes to src/action.py for instance.

LegoChicken, 2017-07-13:
No, that isn't the only service. I discovered this whilst trying to disable the ready led flashing using the instructions in issue #15. See https://github.com/google/aiyprojects-raspbian/tree/master/systemd for a list of all the services.

sheridat, 2017-07-13:
Crikey - well in that case count me in as a +1

ensonic, 2017-07-13:
Personally I think such a script is overkill. We have 3 main services (some of the files in that dir are triggers for boot up and shutdown) and things there aren't changed that often.

LegoChicken, 2017-07-13:
Look at the files in https://github.com/google/aiyprojects-raspbian/tree/master/systemd it appears that the 3 services that should be restarted when the code is updated are:
status-led
status-monitor
voice-recognizer
I think that a short script would be useful for when you are modifying the code. It would make the process of tinkering quicker.

drigz, 2017-10-18:
This is obsolete now that the status LED is no longer a separate service.

closed 2017-10-18 by drigz