google/aiyprojects-raspbian #98: Doesn't work with usb sound cards - cannot pass configuration to Player

ywonglanecove, 2017-07-15:
I tried 3 USB sound cards they all have the same problem

It died when it spoke, because most USB sound cards don't have a hardware mixer, thay can't play mono nor be used by multiple applications at the same time

`[2017-07-15 11:23:37,411] INFO:main:Playing 15.3237 seconds of audio...
aplay: set_params:1233: Sample format non available
Available formats:
- FLOAT_LE
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.4/threading.py", line 920, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.4/threading.py", line 868, in run
    self._target(*self._args, **self._kwargs)
  File "src/main.py", line 374, in _recognize
    self._handle_result(self.recognizer.do_request())
  File "src/main.py", line 392, in _handle_result
    self._play_assistant_response(result.response_audio)
  File "src/main.py", line 404, in _play_assistant_response
    sample_rate=sample_rate_hz)
  File "/home/pi/voice-recognizer-raspi/src/aiy/_drivers/_player.py", line 52, in play_bytes
    aplay.stdin.write(audio_bytes)
BrokenPipeError: [Errno 32] Broken pipe
`
The solution is creating a dmix virtual sound card called dmixed then use -D plug:dmixed for aplay
However main.py doesn't pass its configuration to _player.py so I am hard coding output_device for the time being

I can create a proper PR in a couple of weeks

ywonglanecove, 2017-07-15:
TTS is broken as well 

> play FAIL formats: can't open output file `default': select_format error: Operation not permitted

ywonglanecove, 2017-07-15:
Any reason AIY uses 2 different players, aplay for assistant, sox for tts?

drigz, 2017-07-17:
> then use -D plug:dmixed for aplay

Can't you just change the default device in the ALSA config? This is how we did it for the Voice HAT: https://github.com/google/aiyprojects-raspbian/blob/master/scripts/asound.conf#L21

> Any reason AIY uses 2 different players, aplay for assistant, sox for tts?

Yep - the pico TTS engine generates clipped audio output, which triggers this bug in speexdsp: https://github.com/xiph/speexdsp/issues/13

drigz, 2017-09-05:
@ywonglanecove Did you manage to get an ALSA config with a working default device?

Assuming configuring the default device is enough to make it work, we can just remove the input/output device options rather than making them work again, so I've created #127 for this.

drigz, 2017-10-18:
Closing due to inactivity.

closed 2017-10-18 by drigz