google/aiyprojects-raspbian #101: Improve TTS voice

sheridat, 2017-07-18:
In the Raspberry.org AIY forum we have been discussing the TTS voice and how to improve it 
The thread is https://www.raspberrypi.org/forums/viewtopic.php?f=114&t=188515

A significant improvement is changing tts.py **from** 
try:
        subprocess.call(['pico2wave', '--lang', lang, '-w', raw_wav, words])
        subprocess.call(['play', raw_wav, '--no-show-progress', '--guard'])
    finally:
        os.unlink(raw_wav)
    
**to** 
try:
       **words = '<volume level=\'60\'><pitch level=\'133\'>' + words + '</pitch></volume>'**
        subprocess.call(['pico2wave', '--lang', lang, '-w', raw_wav, words])
        subprocess.call(['play', raw_wav, '--no-show-progress', '--guard'])
    finally:
        os.unlink(raw_wav)

drigz, 2017-07-18:
You guys are so amazing! I tried to reduce the volume with pico2wave but couldn't figure it out.

I'll try this out on a few Voice Kits I have here, but it looks like a good approach. Would you like to create a pull request for this, or should I?

sheridat, 2017-07-18:
Hi, one of the contributors suggests that new line should be better expressed as 
words = '<volume level=\'60\'><pitch level=\'133\'>' + words + '</pitch></volume>'

I haven't got a clue about pull requests so grateful if you would. Thank You

drigz, 2017-07-18:
If you want to insert code into a GitHub comment you can use the button "<>" above the text field, or surround it with ``` like this:

    ```
    words = '<volume level=\'60\'><pitch level=\'133\'>' + words + '</pitch></volume>'
    ```

closed 2017-07-21 by drigz

divx118, 2017-11-18:
I know this is a closed issue, but wouldn't it be nice to expose the volume and pitch like we do with the language as optional arguments. This way we would have some more control in our own code. Since I don't use the voicehat kit, but my own speakers, amplifier and usb soundcard the volume and pitch are still a bit low. It also could give a bit more "emotion" to the voice when responding.
Let me know if this is of interest or not, so if it is worth to making a pull request.

sheridat, 2017-11-19:
Yes – I would also like this. Currently I amend _tts.py as follows
words = '<volume level="60"><pitch level="90">%s</pitch></volume>' % words
rather than the default pitch of 130.

I really do my best to avoid altering any of your modules and much prefer it when you expose arguments that we can set in our own code such as the optional facility to set the language in aiy.audio.say('some words to say', lang='en-GB')

From: Maurice van Kruchten
Sent: 18 November 2017 10:18
To: google/aiyprojects-raspbian
Cc: sheridat; Author
Subject: Re: [google/aiyprojects-raspbian] Improve TTS voice (#101)

I know this is a closed issue, but wouldn't it be nice to expose the volume and pitch like we do with the language as optional arguments. This way we would have some more control in our own code. Since I don't use the voicehat kit, but my own speakers, amplifier and usb soundcard the volume and pitch are still a bit low. It also could give a bit more "emotion" to the voice when responding.
Let me know if this is of interest or not, so if it is worth to making a pull request.

PR #105 (2017-07-21): Improve TTS quality
This was discovered on the MagPi forums. Fixes #101.
commit 2a0f213e (2017-07-21): Improve TTS quality (#105)

This was discovered on the MagPi forums. Fixes #101.