google/aiyprojects-raspbian #114: What sample library am I running

sheridat, 2017-07-31:
Hi, this is a simple question. I've raised a couple of issues with the assistant repo folk such as this [](https://github.com/googlesamples/assistant-sdk-python/issues/88#issuecomment-318950615 )

@proppy has asked me to confirm which sample I am running. I am guessing I am using google-assistant-grpc==0.0.2 but I don't really know. I am using whatever gets called when you use ok-google as the trigger.

Can you please advise me how to answer proppy's question. Thanks

ensonic, 2017-07-31:
According to
https://github.com/google/aiyprojects-raspbian/blob/master/scripts/install-deps.sh
we're pinning google-assistant-library==0.0.2

I see that there is a 0.0.3, I'll check the changes and send a PR to upgrade, but you can also try this locally:

    env/bin/pip install --upgrade google-assistant-library==0.0.3

sheridat, 2017-07-31:
@ensonic Thank You - I did upgrade and hey ho there is an improvement -  if I say good afternoon it no longer at the end says "here is the latest news" followed by silence

sheridat, 2017-08-23:
@ensonic Hi, post no 2 above - did you ever check out the google-assistant-library 0.0.3 changes with a view to changing the install_deps.sh script to move everyone up to 0.0.3?

PR #122 (2017-08-28): Upgrade to google-assistant-library==0.0.3
Fixes #114.
commit 67cba31e (2017-08-28): Upgrade to google-assistant-library==0.0.3 (#122)

Fixes #114.
closed 2017-08-28 by drigz