google/aiyprojects-raspbian #115: Voice Kit

inobmaz, 2017-08-03:
Newbie trying to get the voice kit to work.  I get the following message: 

File "src/main.py", line 300, in <module>
    main()
  File "src/main.py", line 149, in main
    os.path.expanduser(args.assistant_secrets))
  File "src/main.py", line 98, in try_to_get_credentials
    client_secrets, scopes=[ASSISTANT_OAUTH_SCOPE])
  File "/home/pi/voice-recognizer-raspi/env/lib/python3.4/site-packages/googlesamples/assistant/auth_helpers/__init__.py", line 43, in credentials_flow_interactive
    scopes=scopes)
  File "/home/pi/voice-recognizer-raspi/env/lib/python3.4/site-packages/google_auth_oauthlib/flow.py", line 164, in from_client_secrets_file
    with open(client_secrets_file, 'r') as json_file:
IsADirectoryError: [Errno 21] Is a directory: '/home/pi/assistant.json'

Thoughts?

ensonic, 2017-08-03:
Did you follow the setup guide?
https://aiyprojects.withgoogle.com/voice/#users-guide-1-2--turn-on-the-google-assistant-api

inobmaz, 2017-08-03:
Yes, I am going to reinstall the image.  See what happens...

eduncan911, 2017-08-08:
One thing to note was that the MagPi article had a typo in the instructions, and the online users' guide had corrected additional steps - as well as debugging.

drigz, 2017-08-08:
@inobmaz Did you make any progress?

It looks from the error like you created a directory called `assistant.json`. `assistant.json` should be the file that you download from the Google Cloud Platform website (originally called something like `client_secrets_XXXX.json`, not a directory.

inobmaz, 2017-08-08:
Magpi 57 instructions differ from Google's site.  Figured it out.  User error, I didn't have both files in the the same directory. Duh

closed 2017-08-08 by drigz