google/aiyprojects-raspbian #116: How to turn ON/OFF two GPIO at the same time ?

sckam, 2017-08-08:
Dear all experts,

I have my AIY Project.
I can speak to turn ON/OFF one GPIO pin at a time.
But I want also to turn ON/OFF two or more GPIO pins at the same time (example 'on everything' or 'off everything')
I have try four methods below, but all does not able to turn ON/OFF two GPIO at the same time:
In my code below, I am using GPIO17(to connect to a 5VDC Fan) and GPIO27(to connect to a light bulb via relay).

Method 1:
**chan_list = (17,27)
actor.add_keyword('on everything', GpioWrite(17 & 27, True))
actor.add_keyword('off everything', GpioWrite(17 & 27 False))**

#or

**actor.add_keyword('on everything', GpioWrite(27 & 17, True))
actor.add_keyword('off everything', GpioWrite(27 & 17 False))**

#When I say 'on everything' or 'off everything' for the above method, only GPIO17 is response.

Method 2:
**chan_list = (17,27)
actor.add_keyword('turn on all device',  GPIO.output(chan_list, GPIO.HIGH)) 
actor.add_keyword('turn off all device', GPIO.output(chan_list, GPIO.LOW))**  
#When I say 'turn on all device', I will have the following error:

INFO:speech:transcript: turn on all device
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.4/threading.py", line 920, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.4/threading.py", line 868, in run
    self._target(*self._args, **self._kwargs)
  File "src/main.py", line 269, in _recognize
    self._handle_result(self.recognizer.do_request())
  File "src/main.py", line 279, in _handle_result
    if result.transcript and self.actor.handle(result.transcript):
  File "/home/pi/voice-recognizer-raspi/src/actionbase.py", line 42, in handle
    if handler.handle(command):
  File "/home/pi/voice-recognizer-raspi/src/actionbase.py", line 60, in handle
    self.action.run(command)
AttributeError: 'NoneType' object has no attribute 'run'

Method 3:
**chan_list = (17,27)
actor.add_keyword('turn on everything',  GPIO.output(chan_list, True)) # all on
actor.add_keyword('turn off everything',  GPIO.output(chan_list, False))# all off** 
#When I say 'turn on all device', I will have the following error:

INFO:speech:transcript: turn on everything
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.4/threading.py", line 920, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.4/threading.py", line 868, in run
    self._target(*self._args, **self._kwargs)
  File "src/main.py", line 269, in _recognize
    self._handle_result(self.recognizer.do_request())
  File "src/main.py", line 279, in _handle_result
    if result.transcript and self.actor.handle(result.transcript):
  File "/home/pi/voice-recognizer-raspi/src/actionbase.py", line 42, in handle
    if handler.handle(command):
  File "/home/pi/voice-recognizer-raspi/src/actionbase.py", line 60, in handle
    self.action.run(command)
AttributeError: 'NoneType' object has no attribute 'run'

Method 4:
**actor.add_keyword('turn on everything', GpioWrite(17, True))
actor.add_keyword('turn on everything', GpioWrite(27, True))**
#When I say 'turn on everything', only GPIO17 have response.

#or
**actor.add_keyword('turn on everything', GpioWrite(27, True))
actor.add_keyword('turn on everything', GpioWrite(17, True))**
#When I say 'turn on everything', only GPIO27 have response.

Seek expert out there to give advise and provide some sample.

Thank you in advance.

drigz, 2017-08-08:
@sckam, you will have to create a new action, eg `MultiGpioWrite` by copying the `GpioWrite` action. This could take a list of GPIOs to write to, then you use it like:

`actor.add_keyword('turn on everything', GpioWrite([17, 27], True))`

In the `MultiGpioWrite` action's `run` method, you'll use a Python `for` loop to iterate over the list of GPIOs and write to each one.

sckam, 2017-08-08:
Hi drigz,
Is something like these ?

**#!/usr/bin/python
import RPi.GPIO as GPIO
import time
GPIO.setmode(GPIO.BCM)

outputs_led = (2,3,4,5,6,7,8,9)

patterns = {1:(0,1,0,1,0,1,0,1), 2:(1,0,1,0,1,0,1,0)}

for led in outputs_led:
    GPIO.setup(led, GPIO.OUT)
    GPIO.output(led, 0)

try:
    while True:
        for i in 1,2:
            for j in range(8):
                GPIO.output(outputs_led[j], patterns[i][j])
            time.sleep(0.5)

finally:
    GPIO.cleanup()**

drigz, 2017-09-05:
Yep, you'll have to put something like that inside a `run` function in action.py. I'm going to close this issue as this isn't the right place for support requests - if you still need help you should post on the [AIY Forums](https://www.raspberrypi.org/forums/viewforum.php?f=114&sid=3777797ed9706b9ec1351b95b976e9aa).

closed 2017-09-05 by drigz