google/aiyprojects-raspbian #117: Erratic response - Security Handshake failed

sheridat, 2017-08-09:
Hi, my AIY box has been erratic all day.
I can say "good afternoon" and wait up to 10 seconds for a response. If I check the journal this is what I find.
```
Aug 09 14:47:58 raspberrypi python3[869]: [2017-08-09 14:47:58,967] INFO:main:listening...
Aug 09 14:48:00 raspberrypi python3[869]: E0809 14:48:00.318207506    1263 handshake.c:128]            Security handshake failed: {"created":"@1502286480.318067350","description":"Handshake read failed","file":"../../third_party/grpc/src/core/lib/security/transport/handshake.c","file_line":237,"referenced_errors":[{"created":"@1502286480.318049590","description":"FD shutdown","file":"../../third_party/grpc/src/core/lib/iomgr/ev_epoll_linux.c","file_line":1045}]}
Aug 09 14:48:10 raspberrypi python3[869]: [2017-08-09 14:48:10,000] INFO:root:ON_END_OF_UTTERANCE
Aug 09 14:48:10 raspberrypi python3[869]: [2017-08-09 14:48:10,001] INFO:main:thinking...
Aug 09 14:48:10 raspberrypi python3[869]: [2017-08-09 14:48:10,002] INFO:root:ON_RECOGNIZING_SPEECH_FINISHED:
Aug 09 14:48:10 raspberrypi python3[869]: {'text': 'good afternoon'}
Aug 09 14:48:10 raspberrypi python3[869]: [2017-08-09 14:48:10,013] INFO:root:ON_RESPONDING_STARTED:
Aug 09 14:48:10 raspberrypi python3[869]: {'is_error_response': False}
Aug 09 14:48:25 raspberrypi python3[869]: [2017-08-09 14:48:25,070] INFO:root:ON_RESPONDING_FINISHED
Aug 09 14:48:25 raspberrypi python3[869]: [2017-08-09 14:48:25,076] INFO:root:ON_CONVERSATION_TURN_FINISHED:
Aug 09 14:48:25 raspberrypi python3[869]: {'with_follow_on_turn': False}
```
My first reaction was that my SD card might be a fault - so I re-imaged onto another one. That wasn't a cure.

drigz, 2017-08-09:
I can't reproduce this. How reliably does it happen for you? Are you experiencing any trouble with the internet connection on your other devices?

sheridat, 2017-08-09:
Hi, I'm not having any issues with the internet using anything else. Sometimes the answer from the box when it has to phone home for the answer can take an unusually long time-say 10 seconds or more - for something that normally comes back in a couple of seconds. 
I've checked my broadband using my laptop - getting download speed of 120mbs and 13 mbs upload

sheridat, 2017-08-09:
Hi: I installed speedtest-cli on my AIY box
The result
Retrieving speedtest.net configuration...
Testing from Virgin Media (86.31.255.144)...
Retrieving speedtest.net server list...
Selecting best server based on ping...
Hosted by Solway Communications (Carlisle) [90.44 km]: 1189.313 ms
Testing download speed................................................................................
**Download: 29.78 Mbit/s**
Testing upload speed......................................................................................................
**Upload: 11.92 Mbit/s**

drigz, 2017-08-09:
1.2 seconds is a large ping time - are you seeing similar latency on your laptop?

thegnnu, 2017-08-09:
I am also on Virgin if you are on optic fibre not a old BT line also if you are inside the Pixel Desktop
try a ping to 8.8.8.8  but from the command line outside of Desktop. Why is your Virgin going 90Km  to (Carlisle) [90.44 km] surely the have a server closer to you.
I have just tried the ping time 6pm busy period in Bristol for Google and here is  my results.
root@PIAIY:~# ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=58 time=22.7 ms
64 bytes from 8.8.8.8: icmp_seq=2 ttl=58 time=14.4 ms
64 bytes from 8.8.8.8: icmp_seq=3 ttl=58 time=13.3 ms
64 bytes from 8.8.8.8: icmp_seq=4 ttl=58 time=13.6 ms
64 bytes from 8.8.8.8: icmp_seq=5 ttl=58 time=16.3 ms
64 bytes from 8.8.8.8: icmp_seq=6 ttl=58 time=18.5 ms
64 bytes from 8.8.8.8: icmp_seq=7 ttl=58 time=16.5 ms
64 bytes from 8.8.8.8: icmp_seq=8 ttl=58 time=16.2 ms
64 bytes from 8.8.8.8: icmp_seq=9 ttl=58 time=33.0 ms
mostly down in the 18ms and that is using a wifi dongle on a Pi 2B.
Virgin use this site for speed testing  http://beta.speedtest.net/
it has just given me these server that can be used for my location
Bristol Virgin Media
Wells Kierans Ltd
Gloucester Fasthosts Internet
Newport Hub Network Services Ltd
Cardiff Spectrum Internet
Newbury Vodafone UK
Studley zero.net.uk Ltd
Wells Kierans Ltd
most fairly close as the crow flys

TerryR

sheridat, 2017-08-09:
Hi,  I rebooted my router and the results are now
pi@raspberrypi:~ $ speedtest-cli
Retrieving speedtest.net configuration...
Testing from Virgin Media (86.31.255.144)...
Retrieving speedtest.net server list...
Selecting best server based on ping...
Hosted by Bytemark Hosting (York) [121.29 km]: 29.235 ms
Testing download speed................................................................................
Download: 17.88 Mbit/s
Testing upload speed......................................................................................................
Upload: 12.30 Mbit/s
pi@raspberrypi:~ $

I then pinged the google dns server
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=57 time=18.6 ms
64 bytes from 8.8.8.8: icmp_seq=2 ttl=57 time=21.5 ms
64 bytes from 8.8.8.8: icmp_seq=3 ttl=57 time=21.5 ms
64 bytes from 8.8.8.8: icmp_seq=4 ttl=57 time=18.5 ms
64 bytes from 8.8.8.8: icmp_seq=5 ttl=57 time=34.2 ms
64 bytes from 8.8.8.8: icmp_seq=6 ttl=57 time=21.7 ms
64 bytes from 8.8.8.8: icmp_seq=7 ttl=57 time=18.0 ms
64 bytes from 8.8.8.8: icmp_seq=8 ttl=57 time=21.7 ms
64 bytes from 8.8.8.8: icmp_seq=9 ttl=57 time=29.9 ms
64 bytes from 8.8.8.8: icmp_seq=10 ttl=57 time=22.2 ms
64 bytes from 8.8.8.8: icmp_seq=11 ttl=57 time=24.5 ms
64 bytes from 8.8.8.8: icmp_seq=12 ttl=57 time=22.2 ms
64 bytes from 8.8.8.8: icmp_seq=13 ttl=57 time=22.4 ms
64 bytes from 8.8.8.8: icmp_seq=14 ttl=57 time=20.6 ms
64 bytes from 8.8.8.8: icmp_seq=15 ttl=57 time=18.9 ms
64 bytes from 8.8.8.8: icmp_seq=16 ttl=57 time=18.4 ms
64 bytes from 8.8.8.8: icmp_seq=17 ttl=57 time=21.7 ms
64 bytes from 8.8.8.8: icmp_seq=18 ttl=57 time=17.4 ms
64 bytes from 8.8.8.8: icmp_seq=19 ttl=57 time=21.9 ms

so I am assuming that rebooting the router did the trick.

thegnnu, 2017-08-09:
Seems better on the ping but it moved you further away and a slower download speed.
have a look with the Laptop at the test site used by Virgin and see what server are near you. 
Terryr

sheridat, 2017-08-09:
I've used my laptop with the test site used by Virgin and it selects York as the optimal server - I get on my laptop a ping time of 15ms, download 137mbs and upload of 12.04mbs. As usual with Virgin cable it is very slick and fast.  

Using my laptop - I changed the server to one within ten miles and the results were virtually the same - tried another one within ten miles and again the results were similar. 

The problem I was having has disappeared so I am a happy camper. Thanks for the advice that my connection was the issue

closed 2017-08-09 by sheridat