google/aiyprojects-raspbian #119: Not working with Raspbian Stretch

t1m0thyj, 2017-08-18:
I updated my version of Raspbian to Debian 9 (Stretch) which was just released. When I run the "install-deps.sh" script, I get an error saying:
`Package libttspico-utils is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or is only available from another source

E: Package 'libttspico-utils' has no installation candidate`
It appears that the "libttspico" package is currently only available for i386 architecture, and not the Pi which is ARM. Is there a way to fix this?

drigz, 2017-08-18:
Thanks for the heads up! I've not tried this on Stretch, so I'll look into this.

t1m0thyj, 2017-08-19:
I figured out a way to work around this issue for now:
First, I installed libttspico-data using apt-get in Raspbian.
Then, I downloaded the armhf versions of the packages libttspico0 and libttspico-utils from the Debian Stretch package index:
https://packages.debian.org/stretch/armhf/libttspico0/download
https://packages.debian.org/stretch/armhf/libttspico-utils/download
They must be installed in that order, because the first package is a dependency of the second one.
Finally, the install-deps.sh script gave an error saying that virtualenv could not be found. I fixed it by running "sudo /usr/bin/easy_install virtualenv" as suggested [here](sudo /usr/bin/easy_install virtualenv).

t1m0thyj, 2017-08-19:
I created a pull request that fixes the second issue (the virtualenv one), but not the first one (with the libttspico-utils package). I would fix the first one too, but don't know how to because it seems like it requires adding packages to the official Raspbian repository.

drigz, 2017-08-21:
@t1m0thyj I've sent an email to the Raspbian team to ask about adding those packages to the upstream repo, and I'm adding a warning for now.

Sispheor, 2017-09-04:
Hi, the bug is still there today (04-09-2017).

I've followed your workaround to install the lib anyway. Do you have any news from the Raspbian team?

drigz, 2017-09-04:
Nope, no update. Just sent another email to chase it up. We should probably adapt the instructions to include the workaround if this isn't going to be changed upstream.

t1m0thyj, 2017-09-04:
If the workaround is still required, would it be worth checking the version of Debian in the install-deps.sh script as is mentioned [here](https://github.com/ev3dev/ev3dev/issues/930)? If Stretch is installed, the script could grab the Debian packages and install them to automate the workaround.

drigz, 2017-09-04:
Simon at Raspbian just let me know they're going to deal with it, so this should be resolved in a few days.

t1m0thyj, 2017-09-04:
I just ran apt-get update and the packages are in the official Raspbian Stretch repository for me now.

closed 2017-09-04 by t1m0thyj

Sispheor, 2017-09-04:
I confirm @t1m0thyj . Thank you guys.