google/aiyprojects-raspbian #129: Service fails if Retropie gets installed

dehein, 2017-09-08:
I tried to use both Google aiy and Retropie on the same raspberry pi 3. But somehow the service fails and I can't trigger the ok-google command.

My error message:
    pi@powerpi3 ~> sudo systemctl status voice-recognizer.service -l
● voice-recognizer.service - voice recognizer
   Loaded: loaded (/lib/systemd/system/voice-recognizer.service; enabled)
   Active: active (running) since Fr 2017-09-08 13:01:37 CEST; 8min ago
 Main PID: 530 (python3)
   CGroup: /system.slice/voice-recognizer.service
           └─530 /home/pi/voice-recognizer-raspi/env/bin/python3 -u src/main.py

Sep 08 13:01:37 powerpi3 systemd[1]: Starting voice recognizer...
Sep 08 13:01:37 powerpi3 systemd[1]: Started voice recognizer.
Sep 08 13:01:44 powerpi3 python3[530]: ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to open slave
Sep 08 13:03:07 powerpi3 python3[530]: [530:1253:ERROR:audio_input_processor.cc(755)] Input error
Sep 08 13:03:07 powerpi3 python3[530]: [2017-09-08 13:03:07,264] INFO:root:ON_MUTED_CHANGED:
Sep 08 13:03:07 powerpi3 python3[530]: {'is_muted': False}
Sep 08 13:03:07 powerpi3 python3[530]: [2017-09-08 13:03:07,266] INFO:root:ON_START_FINISHED
Sep 08 13:03:07 powerpi3 python3[530]: [2017-09-08 13:03:07,268] INFO:main:ready...
Sep 08 13:03:07 powerpi3 python3[530]: [2017-09-08 13:03:07,270] INFO:root:ON_ASSISTANT_ERROR:
Sep 08 13:03:07 powerpi3 python3[530]: {'is_fatal': True}

drigz, 2017-09-08:
AIY uses `arecord` to record audio, but I'm guessing the combination of your ALSA configuration and Retropie stop this from working. You can try running the `check_audio` script - if that also fails, we can't do much to fix this in AIY.

Is Retropie running when AIY fails? If not, Retropie might have changed the ALSA config in /etc/asound.conf (which should be the same as scripts/asound.conf in this repo if you're using the Voice HAT) or ~/.asoundrc (which shouldn't exist if you're using the Voice HAT).

ransonneil, 2018-01-03:
How exactly did you get retropie on?
Retropie is an SD image, so I can't figure out how to get it on, or did you install emulation station and retroarch separately?

dehein, 2018-01-05:
Retropie has a Manual Installation option. You Download the correct github Link and execute the Setup. Here ist will build all your selected Things. But have still Not tried again If the Problem can somehow be fixed. I Changes the Image Back to Something else

ransonneil, 2018-01-07:
I decided to go the easy route and mount retropie to a separate SD card, I've just ordered an SD card switched too which should mount nicely inside the case