google/aiyprojects-raspbian #130: Get two answer with different sounds

Blacx, 2017-09-13:
I get two different answer using different sounds with this conditions:

If I using **GPIO trigger**, the first sounds comes from TTS that sounds stiff as robot. and the second sound comes from real google assistant sound. Here is the log:
```
(env) pi@whisnupi:~/voice-recognizer-raspi $ python src/main.py
[2017-09-13 20:13:26,305] WARNING:main:File /tmp/status-led specified for --led-fifo does not exist.
[2017-09-13 20:13:26,316] INFO:recorder:started recording
[2017-09-13 20:13:26,344] INFO:main:ready...
Press the button on GPIO 23 then speak, or press Ctrl+C to quit...
[2017-09-13 20:13:33,107] INFO:main:listening...
[2017-09-13 20:13:33,109] INFO:main:recognizing...
[2017-09-13 20:13:38,439] INFO:speech:event_type: 1
[2017-09-13 20:13:38,442] INFO:main:thinking...
[2017-09-13 20:13:38,511] INFO:speech:transcript: repeat after me good morning
[2017-09-13 20:13:38,753] ERROR:tts:Using fallback directory for TTS output
Traceback (most recent call last):
  File "/home/pi/voice-recognizer-raspi/src/aiy/_drivers/_tts.py", line 48, in say
    (fd, tts_wav) = tempfile.mkstemp(suffix='.wav', dir=TMP_DIR)
  File "/home/pi/voice-recognizer-raspi/env/lib/python3.4/tempfile.py", line 409, in mkstemp
    return _mkstemp_inner(dir, prefix, suffix, flags)
  File "/home/pi/voice-recognizer-raspi/env/lib/python3.4/tempfile.py", line 339, in _mkstemp_inner
    fd = _os.open(file, flags, 0o600)
FileNotFoundError: [Errno 2] No such file or directory: '/run/user/1000/tmpo6qhg7nk.wav'
[2017-09-13 20:13:39,899] INFO:main:handled local command: repeat after me good morning
[2017-09-13 20:13:39,900] INFO:main:Playing 0.8083 seconds of audio...
[2017-09-13 20:13:40,818] INFO:main:ready...
```
and if I using **ok-google trigger**, it will sound two answers coming from TTS and the google assistant talking simultaneously. Here is the log:
```
(env) pi@whisnupi:~/voice-recognizer-raspi $ python src/main.py
[2017-09-13 20:24:18,692] WARNING:main:File /tmp/status-led specified for --led-fifo does not exist.
[2017-09-13 20:24:19,733] INFO:root:ON_MUTED_CHANGED:
  {'is_muted': False}
[2017-09-13 20:24:19,734] INFO:root:ON_START_FINISHED
[2017-09-13 20:24:19,735] INFO:main:ready...
Say "OK, Google" then speak, or press Ctrl+C to quit...
[2017-09-13 20:24:22,388] INFO:root:ON_CONVERSATION_TURN_STARTED
[2017-09-13 20:24:22,389] INFO:main:listening...
[2017-09-13 20:24:26,160] INFO:root:ON_END_OF_UTTERANCE
[2017-09-13 20:24:26,161] INFO:main:thinking...
[2017-09-13 20:24:26,166] INFO:root:ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'repeat after me good morning'}
[2017-09-13 20:24:26,168] ERROR:tts:Using fallback directory for TTS output
Traceback (most recent call last):
  File "/home/pi/voice-recognizer-raspi/src/aiy/_drivers/_tts.py", line 48, in say
    (fd, tts_wav) = tempfile.mkstemp(suffix='.wav', dir=TMP_DIR)
  File "/home/pi/voice-recognizer-raspi/env/lib/python3.4/tempfile.py", line 409, in mkstemp
    return _mkstemp_inner(dir, prefix, suffix, flags)
  File "/home/pi/voice-recognizer-raspi/env/lib/python3.4/tempfile.py", line 339, in _mkstemp_inner
    fd = _os.open(file, flags, 0o600)
FileNotFoundError: [Errno 2] No such file or directory: '/run/user/1000/tmp48qj0hey.wav'
[2017-09-13 20:24:27,324] INFO:root:ON_RESPONDING_STARTED:
  {'is_error_response': False}
[2017-09-13 20:24:27,325] INFO:root:ON_RESPONDING_FINISHED
[2017-09-13 20:24:27,327] INFO:root:ON_CONVERSATION_TURN_FINISHED:
  {'with_follow_on_turn': False}
[2017-09-13 20:24:27,328] INFO:main:ready...
```

Can you help me to find out what the problem is?

drigz, 2017-09-15:
Is it possible that you have the voice recognizer running as a service? This is normally the case when people get two responses. Try `sudo service voice-recognizer stop` and see if the issue continues.

Otherwise, here is the code for choosing response type - you could add some logging to try to debug it: [gpio trigger](https://github.com/google/aiyprojects-raspbian/blob/8024b60cd7a8c7536e2a2936f5f082e97ffdce89/src/main.py#L387), [ok-google trigger](https://github.com/google/aiyprojects-raspbian/blob/8024b60cd7a8c7536e2a2936f5f082e97ffdce89/src/main.py#L221)

Blacx, 2017-09-27:
@drigz I'm still get two sound replaying my local actor "Repeate after me" or "Ip address".
and i'm get this information when I execute `sudo service voice-recognizer stop` command.
=>
`Failed to stop voice-recognizer.service: Unit voice-recognizer.service not loaded.
`

ensonic, 2017-09-28:
Can you check which programs are running?
`ps -ef | grep env/bin/python3` - this should be zero or one, but not more than one.

closed 2018-06-29 by dmitriykovalev