google/aiyprojects-raspbian #132: service unavailible

aycgit, 2017-09-15:
when i was looking at triggers, i got the error:
```
`Traceback (most recent call last):
  File "main.py", line 373, in _recognize
    self._handle_result(self.recognizer.do_request())
  File "/home/pi/voice-recognizer-raspi/src/speech.py", line 259, in do_request
    return self._handle_response_stream(response_stream)
  File "/home/pi/voice-recognizer-raspi/src/speech.py", line 207, in _handle_response_stream
    raise Error('Server error: ' + resp.error.message)
speech.Error: Server error: Service unavailable.
`
```
although i don't think it is relevant: my test weirdly recognized sound at start -> thinks -> this error spoken to the speaker -> ready -> my thing listens but callback can't be called anymore

aycgit, 2017-09-15:
my (very unclean) code: [gist](https://gist.github.com/aycgit/43bb7e7ec56180b058e6400ff87caca6)

drigz, 2017-09-19:
I guess this is a temporary error in the backend. Did you see it repeatedly or just once? Are you still seeing it?

Also, could you explain further what you mean by "my thing listens but callback can't be called anymore"? In particular, which callback are you referring to?

drigz, 2017-10-18:
Closing due to inactivity.

closed 2017-10-18 by drigz