google/aiyprojects-raspbian #133: executables breaking on ImportError after fresh install of latest Raspbian

zizipoil, 2017-09-16:
Hello, 

I'm having issues running the different executables present in /src
They all seem to return import errors.

### config
I use a raspberry pi 3b.
I use the latest Raspbian 'Strech with desktop' image (Release date:2017-09-07, Kernel version:4.9)
I cloned the main branch of this repo and followed the steps written in _HACKING.md_. No errors to report at this stage.

I am not using the voice hat, so I use my own version of _/etc/asound.conf_ to use the jack as output and a USB mic.

### tests
Checkpoints _check_audio.py_, _check_cloud.py_ and _check_wifi.py_ all work fine.
Checkpoint _load_test.py_ fails (see Traces)

# Traces
These are the traces I get for the different executables of the _src_ folder:
### main.py
```bash
pi@raspberrypi:~/voice-recognizer-raspi $ src/main.py 
Traceback (most recent call last):
  File "src/main.py", line 26, in <module>
    import configargparse
ImportError: No module named 'configargparse'
```
### cloudspeech_demo.py
```bash
pi@raspberrypi:~/voice-recognizer-raspi $ src/cloudspeech_demo.py 
Traceback (most recent call last):
  File "src/cloudspeech_demo.py", line 21, in <module>
    import aiy.cloudspeech
  File "/home/pi/voice-recognizer-raspi/src/aiy/cloudspeech.py", line 19, in <module>
    import aiy._apis._speech
  File "/home/pi/voice-recognizer-raspi/src/aiy/_apis/_speech.py", line 24, in <module>
    import google.auth
ImportError: No module named 'google'
```
### assistant_grpc_demo.py
```bash
pi@raspberrypi:~/voice-recognizer-raspi $ src/assistant_grpc_demo.py 
Traceback (most recent call last):
  File "src/assistant_grpc_demo.py", line 21, in <module>
    import aiy.assistant.grpc
  File "/home/pi/voice-recognizer-raspi/src/aiy/assistant/grpc.py", line 19, in <module>
    import aiy._apis._speech
  File "/home/pi/voice-recognizer-raspi/src/aiy/_apis/_speech.py", line 24, in <module>
    import google.auth
ImportError: No module named 'google'
```
### assistant_library_demo.py
```bash
pi@raspberrypi:~/voice-recognizer-raspi $ src/assistant_library_demo.py 
Traceback (most recent call last):
  File "src/assistant_library_demo.py", line 30, in <module>
    import aiy.assistant.auth_helpers
  File "/home/pi/voice-recognizer-raspi/src/aiy/assistant/auth_helpers.py", line 24, in <module>
    import google_auth_oauthlib.flow
ImportError: No module named 'google_auth_oauthlib'
```
### assistant_library_with_button_demo.py 
```bash
pi@raspberrypi:~/voice-recognizer-raspi $ src/assistant_library_with_button_demo.py 
Traceback (most recent call last):
  File "src/assistant_library_with_button_demo.py", line 31, in <module>
    import aiy.assistant.auth_helpers
  File "/home/pi/voice-recognizer-raspi/src/aiy/assistant/auth_helpers.py", line 24, in <module>
    import google_auth_oauthlib.flow
ImportError: No module named 'google_auth_oauthlib'
```
### load_test.py
```bash
pi@raspberrypi:~/voice-recognizer-raspi $ ./checkpoints/load_test.py 
Running test forever - press Ctrl+C to stop...
recording  Traceback (most recent call last):
  File "./checkpoints/load_test.py", line 181, in <module>
    main()
  File "./checkpoints/load_test.py", line 173, in main
    run_test()
  File "./checkpoints/load_test.py", line 145, in run_test
    record_wav()
  File "./checkpoints/load_test.py", line 121, in record_wav
    stdout=subprocess.DEVNULL, stderr=subprocess.DEVNULL)
  File "/usr/lib/python3.5/subprocess.py", line 271, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/pi/voice-recognizer-raspi/env/bin/python3', '/home/pi/voice-recognizer-raspi/src/aiy/audio.py', 'dump', '/tmp/tmpfwzdt9he.wav', '-d', '3']' returned non-zero exit status 1
Press Enter to close...`
```

# my 2 cents
As you can see all executables fail with import errors. 
I'm suspecting that _/scripts/install-deps.sh_ is somehow creating the error.

Thanks in advance for your help
Raph

zizipoil, 2017-09-16:
Just realised that running 
```bash
cd ~/voice-recognizer-raspi
source env/bin/activate
python3 src/main.py
```
prevents these errors form happening.
But is that the way things are supposed to be ran? (aka in a virtual env)

sheridat, 2017-09-16:
Yes it is meant to be run in it's own python virtual environment.

zizipoil, 2017-09-21:
Ok, I'll close the issue then.
Thanks for your answer

closed 2017-09-21 by zizipoil