google/aiyprojects-raspbian #134: Problems with Sept 11th image

sheridat, 2017-09-20:
Hi, There are reports on the raspberry.org aiy forum regarding issues with the latest AIY image - please see this thread [](https://www.raspberrypi.org/forums/viewtopic.php?f=114&t=193379&sid=54b56399b5653a3bc62fe52715cb1b0d)
Can you advise?

drigz, 2017-09-20:
Thanks for passing this through to us, @sheridat! We'll look into it and see if we need to update the docs or the image.

zizipoil, 2017-09-21:
As a noob, what I found downloading and using the image was that it seems to be based on the voiceKit branch when most of the docs (and other blog posts I read about it) referred to the code of the master branch.

drigz, 2017-09-28:
The instructions have now been updated for the new image.

closed 2017-09-28 by drigz