google/aiyprojects-raspbian #136: Connection Errors whenever PI AIY is rebooted

sheridat, 2017-09-27:
Hi, 
I've been doing some new integrations with my home automation software (home assistant) over the last couple of days and have therefore been checking the AIY logs frequently. My voice recognizer service is loaded automatically upon boot/reboot. 
I've noticed that every time I reboot a bunchload of error messages are thrown out whilst the AIY service is being established.  It then fails then restarts itself and then we are ready to go.  
In the following traceback see the period starting 

-- Logs begin at Wed 2017-09-27 10:36:05 BST. --
Sep 27 10:36:08 AIY systemd[1]: Starting voice recognizer...

**thru to** 

[Sep 27 10:36:14 AIY systemd[1]: voice-recognizer.service: main process exited, code=exited, status=1/FAILURE
Sep 27 10:36:14 AIY systemd[1]: Unit voice-recognizer.service entered failed state.
Sep 27 10:36:14 AIY systemd[1]: voice-recognizer.service holdoff time over, scheduling restart.
Sep 27 10:36:14 AIY systemd[1]: Stopping voice recognizer...
Sep 27 10:36:14 AIY systemd[1]: Starting voice recognizer...
Sep 27 10:36:14 AIY systemd[1]: Started voice recognizer.
Once it has restarted itself everything is fine as you can see in the rest of the traceback where I switch some lights on and then off
As a precursor I've checked my internet connection and the results are fine
pi@AIY:~ $ speedtest-cli
Retrieving speedtest.net configuration...
Testing from Virgin Media (86.31.253.20)...
Retrieving speedtest.net server list...
Selecting best server based on ping...
Hosted by Bytemark Hosting (York) [119.40 km]: 38.129 ms
Testing download speed................................................................................
Download: 35.40 Mbit/s
Testing upload speed......................................................................................................
Upload: 12.14 Mbit/s

Traceback
pi@AIY:~ $ sudo journalctl -u voice-recognizer -n 100 -f
-- Logs begin at Wed 2017-09-27 10:36:05 BST. --
Sep 27 10:36:08 AIY systemd[1]: Starting voice recognizer...
Sep 27 10:36:08 AIY systemd[1]: Started voice recognizer.
Sep 27 10:36:13 AIY python3[562]: Traceback (most recent call last):
Sep 27 10:36:13 AIY python3[562]: File "/home/pi/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/connection.py", line 141, in _new_conn
Sep 27 10:36:13 AIY python3[562]: (self.host, self.port), self.timeout, **extra_kw)
Sep 27 10:36:13 AIY python3[562]: File "/home/pi/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/util/connection.py", line 60, in create_connection
Sep 27 10:36:13 AIY python3[562]: for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):
Sep 27 10:36:13 AIY python3[562]: File "/usr/lib/python3.4/socket.py", line 530, in getaddrinfo
Sep 27 10:36:13 AIY python3[562]: for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
Sep 27 10:36:13 AIY python3[562]: socket.gaierror: [Errno -2] Name or service not known
Sep 27 10:36:13 AIY python3[562]: During handling of the above exception, another exception occurred:
Sep 27 10:36:13 AIY python3[562]: Traceback (most recent call last):
Sep 27 10:36:13 AIY python3[562]: File "/home/pi/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/connectionpool.py", line 600, in urlopen
Sep 27 10:36:13 AIY python3[562]: chunked=chunked)
Sep 27 10:36:13 AIY python3[562]: File "/home/pi/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/connectionpool.py", line 345, in _make_request
Sep 27 10:36:13 AIY python3[562]: self._validate_conn(conn)
Sep 27 10:36:13 AIY python3[562]: File "/home/pi/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/connectionpool.py", line 844, in _validate_conn
Sep 27 10:36:13 AIY python3[562]: conn.connect()
Sep 27 10:36:13 AIY python3[562]: File "/home/pi/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/connection.py", line 284, in connect
Sep 27 10:36:13 AIY python3[562]: conn = self._new_conn()
Sep 27 10:36:13 AIY python3[562]: File "/home/pi/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/connection.py", line 150, in _new_conn
Sep 27 10:36:13 AIY python3[562]: self, "Failed to establish a new connection: %s" % e)
Sep 27 10:36:13 AIY python3[562]: requests.packages.urllib3.exceptions.NewConnectionError: <requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x74996cd0>: Failed to establish a new connection: [Errno -2] Name or service not known
Sep 27 10:36:13 AIY python3[562]: During handling of the above exception, another exception occurred:
Sep 27 10:36:13 AIY python3[562]: Traceback (most recent call last):
Sep 27 10:36:13 AIY python3[562]: File "/home/pi/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/adapters.py", line 438, in send
Sep 27 10:36:13 AIY python3[562]: timeout=timeout
Sep 27 10:36:13 AIY python3[562]: File "/home/pi/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/connectionpool.py", line 649, in urlopen
Sep 27 10:36:13 AIY python3[562]: _stacktrace=sys.exc_info()[2])
Sep 27 10:36:13 AIY python3[562]: File "/home/pi/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/util/retry.py", line 388, in increment
Sep 27 10:36:13 AIY python3[562]: raise MaxRetryError(_pool, url, error or ResponseError(cause))
Sep 27 10:36:13 AIY python3[562]: requests.packages.urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='accounts.google.com', port=443): Max retries exceeded with url: /o/oauth2/token (Caused by NewConnectionError('<requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x74996cd0>: Failed to establish a new connection: [Errno -2] Name or service not known',))
Sep 27 10:36:13 AIY python3[562]: During handling of the above exception, another exception occurred:
Sep 27 10:36:13 AIY python3[562]: Traceback (most recent call last):
Sep 27 10:36:13 AIY python3[562]: File "/home/pi/voice-recognizer-raspi/env/lib/python3.4/site-packages/google/auth/transport/requests.py", line 112, in __call__
Sep 27 10:36:13 AIY python3[562]: **kwargs)
Sep 27 10:36:13 AIY python3[562]: File "/home/pi/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/sessions.py", line 518, in request
Sep 27 10:36:13 AIY python3[562]: resp = self.send(prep, **send_kwargs)
[... 27 lines cut ...]
Sep 27 10:36:14 AIY systemd[1]: Stopping voice recognizer...
Sep 27 10:36:14 AIY systemd[1]: Starting voice recognizer...
Sep 27 10:36:14 AIY systemd[1]: Started voice recognizer.
Sep 27 10:36:18 AIY python3[904]: [2017-09-27 10:36:18,716] INFO:root:ON_MUTED_CHANGED:
Sep 27 10:36:18 AIY python3[904]: {'is_muted': False}
Sep 27 10:36:18 AIY python3[904]: [2017-09-27 10:36:18,721] INFO:root:ON_START_FINISHED
Sep 27 10:36:18 AIY python3[904]: [2017-09-27 10:36:18,722] INFO:main:ready...
Sep 27 10:36:35 AIY python3[904]: [2017-09-27 10:36:35,481] INFO:root:ON_CONVERSATION_TURN_STARTED
Sep 27 10:36:35 AIY python3[904]: [2017-09-27 10:36:35,482] INFO:main:listening...
Sep 27 10:36:37 AIY python3[904]: [2017-09-27 10:36:37,976] INFO:root:ON_END_OF_UTTERANCE
Sep 27 10:36:37 AIY python3[904]: [2017-09-27 10:36:37,978] INFO:main:thinking...
Sep 27 10:36:37 AIY python3[904]: [2017-09-27 10:36:37,990] INFO:root:ON_RECOGNIZING_SPEECH_FINISHED:
Sep 27 10:36:37 AIY python3[904]: {'text': 'switch lights'}
Sep 27 10:36:39 AIY python3[904]: [{"attributes": {"assumed_state": false, "auto": true, "entity_id": ["switch.ck_light", "switch.ts_light", "switch.living_room_light"], "friendly_name": "all switches", "hidden": true, "order": 1}, "entity_id": "group.all_switches", "last_changed": "2017-09-27T09:36:39.520444+00:00", "last_updated": "2017-09-27T09:36:39.520444+00:00", "state": "on"}, {"attributes": {"friendly_name": "living_room_light"}, "entity_id": "switch.living_room_light", "last_changed": "2017-09-27T09:36:39.516729+00:00", "last_updated": "2017-09-27T09:36:39.516729+00:00", "state": "on"}]
Sep 27 10:36:41 AIY python3[904]: [2017-09-27 10:36:41,016] INFO:root:ON_CONVERSATION_TURN_FINISHED:
Sep 27 10:36:41 AIY python3[904]: {'with_follow_on_turn': False}
Sep 27 10:36:41 AIY python3[904]: [2017-09-27 10:36:41,024] INFO:main:ready...
Sep 27 10:36:43 AIY python3[904]: [2017-09-27 10:36:43,113] INFO:root:ON_CONVERSATION_TURN_STARTED
Sep 27 10:36:43 AIY python3[904]: [2017-09-27 10:36:43,115] INFO:main:listening...
[... cut ...] idden": true, "order": 1}, "entity_id": "group.all_switches", "last_changed": "2017-09-27T09:36:46.913713+00:00", "last_updated": "2017-09-27T09:36:46.913713+00:00", "state": "off"}, {"attributes": {"friendly_name": "living_room_light"}, "entity_id": "switch.living_room_light", "last_changed": "2017-09-27T09:36:46.907820+00:00", "last_updated": "2017-09-27T09:36:46.907820+00:00", "state": "off"}]
Sep 27 10:36:48 AIY python3[904]: [2017-09-27 10:36:48,341] INFO:root:ON_CONVERSATION_TURN_FINISHED:

sheridat, 2017-09-27:
Aha,  The PI has a configuration program. One of the items is "Network at boot" and you can tick it to say wait for the network - I ticked it and hey presto all that stuff at the beginning disappears! Hooray

closed 2017-09-27 by sheridat

ensonic, 2017-09-28:
Thanks for this tip!
For the record - this is `raspi-config` and the option is under `3 - Boot Options / B2 Wait for Network at Boot`