google/aiyprojects-raspbian #140: Requirement for clear documentation or a proper wiki

shivasiddharth, 2017-10-02:
Hello,
The documentation of the AIY Projects in the offical [webpage](https://aiyprojects.withgoogle.com/voice/) seems to conflict with master branch. I have been around for quite sometime so its not an issue, but for a newbie or  someone who is visiting the git for first time, it could be perplexing. 
Also if @drigz reply to #138 is mentioned in the readme or hacking.md file, it will throw more light on the existence of two separate branches for the AIY projects.
Regards

drigz, 2017-10-10:
Could you have a look at these pull requests? #144 #145
Do they address the issue you're describing?

shivasiddharth, 2017-10-11:
@drigz , Yes thats what i was talking about.

closed 2017-10-11 by shivasiddharth