google/aiyprojects-raspbian #141: Thank You for providing assistant_library_with_local_commands_demo.py -  response quality

sheridat, 2017-10-05:
A heartfelt Thank You for providing an additional voicekit demo 'assistant_library_with_local_commands_demo.py' showing how we can use the google assistant library to perform local commands as we used to using src/action.py

I've started the conversion process for my local commands and am a much happier bunny!

Yourselves will remember the work you put in to improve the local voice response. The voice kit audio isn't as good as the master voice version - will you be able to improve it?

Thanks again.

mpember, 2017-10-06:
Does anyone know how the 'turns' work?   Does the new structure make it easier to have a multi-part conversation?

e.g., I have code which warns that a podcast about to be played is old.  But if it is old, the only option is to reach for the GPIO button and stop the playback.  It would be ceate to simply ask an "are you sure you still want to play it?" prompt and wait for a "yes / no" answer.

shivasiddharth, 2017-10-06:
@sheridat , I have done this modification for custom actions even for the assistant librray (SDK-Ok Google). https://github.com/shivasiddharth/GassistPi/blob/master/src/main.py hope you find it useful.

mpember, 2017-10-06:
@shivasiddharth I know this is going to sound like I'm being pedantic, but....   Is there a reason that lines 90 and 92 of your gRPC code uses .lower() on a hard-coded string value?  Is there any scenario where they could be anything other than lowercase?

Maybe it is just my mild OCD kicking in.

shivasiddharth, 2017-10-06:
@mpember , Its just to play safe. Users have the option to change the trigger word and commands. If someone uses a proper noun, it could result in the argument having mixed cases ( I have not tested it though). I felt that i would loose nothing by turning all strings to lower case.

drigz, 2017-10-10:
I think there are a couple of different conversations going on here. @sheridat, is this issue about the quality of the TTS audio? (ie from `aiy.audio.say`)

Could you explain more precisely what you mean when you say the audio isn't as good? The new branch has the [volume and pitch tweaks](https://github.com/google/aiyprojects-raspbian/blob/6af695bdceaabdd4144def6a11f930e9d02a2d5c/src/aiy/_drivers/_tts.py#L50) that improved the quality before, so I'd expect it to work well.

sheridat, 2017-10-10:
Your right - apologies for causing the confusion. I will redo some testing later today and get back to you with a more precise description of the problem.

sheridat, 2017-10-10:
To cut a long story short after a bit of messing around with both branches I have isolated the issue to the language setting. You will remember that in the master branch you could set the config file to give you either en-US or en-GB and some of us preferred the GB variant.  I've just gone and changed 
i18n.py's _DEFAULT_LANGUAGE_CODE = 'en-US' to '_DEFAULT_LANGUAGE_CODE = 'en-GB' and that really nice voice has returned.

Now that the config file has disappeared is there a way to set the default language code without mangling i18n.py?

drigz, 2017-10-10:
You should be able to call `aiy.i18n.set_language_code('en-GB')` to set the default language, or use `aiy.audio.say('hello', lang='en-GB')` to change it for an individual response.

sheridat, 2017-10-10:
Great! - Thank you. Tried both - will publicise that on the Raspberry.org AIY forum. Closing issue.

closed 2017-10-10 by sheridat