google/aiyprojects-raspbian #142: VOICEKIT ALSA lib pcm.c:7843:(snd_pcm_recover) overrun occurred

sheridat, 2017-10-06:
Hi, 
Using your new google src/assistant_library_with_local_commands_demo.py. 
I've amended it to turn my TV on and off. 
Whilst testing it I noticed that after a couple of on/off iterations it started throwing out a message 
ALSA lib pcm.c:7843:(snd_pcm_recover) overrun occurred. It does however still turn the TV on and off.

So the message isn't killing things but perhaps there is some sort of problem.

Traceback is pretty self explanatory. If you need any more info - just say

Thanks

Build info: Tue Sep 12 00:36:20 UTC 2017 @ 3955cac
Dev terminal is ready! Hint: enter src/assistant_library_demo.py
(env) pi@raspberrypi:~/AIY-voice-kit-python $ src/assistant_library_with_local_commands_demo.py 
/home/pi/AIY-voice-kit-python/src/aiy/_drivers/_led.py:51: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(channel, GPIO.OUT)
Say "OK, Google" then speak, or press Ctrl+C to quit...
You said: TV on
You said: 
You said: TV on
You said: TV off
ALSA lib pcm.c:7843:(snd_pcm_recover) overrun occurred
You said: TV on
ALSA lib pcm.c:7843:(snd_pcm_recover) overrun occurred
You said: TV off
ALSA lib pcm.c:7843:(snd_pcm_recover) overrun occurred
You said: TV on
ALSA lib pcm.c:7843:(snd_pcm_recover) overrun occurred
You said: TV off
ALSA lib pcm.c:7843:(snd_pcm_recover) overrun occurred
You said: TV on
ALSA lib pcm.c:7843:(snd_pcm_recover) overrun occurred

drigz, 2017-10-09:
Thanks for the report! We've seen this error  before, although less
frequently since we switched from PyAudio to arecord.

Since there are no timestamps, can you give some information on timing? ie
how far apart were the commands, did the error come immediately before or
after the command?

Are you using Stretch or an older Raspbian version?

<https://github.com/google/aiyprojects-raspbian/issues/142>, or mute the
thread

sheridat, 2017-10-09:
Hi, In terms of sequence the errors came immediately after saying the command. 
IN terms of the command there is a 3 to 4 second lag from saying "TV" on before it is visually apparent that it has turned on, whereas saying TV off results in the TV switching of immediately so the timing sequence for the testing I was doing was 

You said: TV on
wait 5 seconds
You said: TV off
error 
1 second 
You said: TV on
error
5 seconds
You said: TV off
error
1 second
You said: TV on
error 
5 seconds
You said: TV off
error 
1 second
You said: TV on

I had tested this a couple of weeks ago using the master branch and never saw this error.
I also substituted the microphone from my other AIY kit and it made no difference.

subterain, 2017-12-16:
Hi, I am also having the same error with similar circumstances. If I run the Assistant straight after booting the pi, it runs fine. But, I have another program that runs alongside the Assistant that looks after RF Tx Rx and various other IO related control using the pigpio library. I have narrowed it down to the point in time where I call gpioInitialise() in my c++ code. After that, the audio through the voice hat is corrupted to a burst of maximum volume white noise, and audio recording through the voice hat mic shows the same error (ALSA lib pcm.c:7843:(snd_pcm_recover) overrun occurred), but it repeats as fast as the terminal window can print. I am a bit of a noob to all this so I can only guess that issue is related to sheridat as in gpioInitialise() is possibly affecting the overall GPIO sampling rate somehow in the same way with whatever IO control that happens immediately after sheridat says "TV on" - or off.

The white noise on my audio output only occurs for the first wave playback after gpioTerminate() is called, then it returns to normal, but the audio recording is corrupted until I reboot the pi - still struggling to find a work-around. If I run the checkpoints/CheckAudio.py script I get the following output:

Playing a test sound...
Did you hear the test sound? (y/n) y
When you're ready, press enter and say 'Testing, 1 2 3'...
Recording...
overrun!!! (at least 18.610 ms long)
overrun!!! (at least 14.154 ms long)
overrun!!! (at least 14.744 ms long)
overrun!!! (at least 6.536 ms long)
overrun!!! (at least 9.530 ms long)
overrun!!! (at least 6.134 ms long)
overrun!!! (at least 7.404 ms long)
overrun!!! (at least 7.202 ms long)
overrun!!! (at least 9.832 ms long)
overrun!!! (at least 7.289 ms long)
overrun!!! (at least 9.869 ms long)
overrun!!! (at least 9.818 ms long)
overrun!!! (at least 7.203 ms long)
overrun!!! (at least 7.603 ms long)
overrun!!! (at least 7.372 ms long)
overrun!!! (at least 7.620 ms long)
overrun!!! (at least 7.527 ms long)
overrun!!! (at least 7.394 ms long)
overrun!!! (at least 7.557 ms long)
overrun!!! (at least 7.529 ms long)
overrun!!! (at least 7.933 ms long)
overrun!!! (at least 9.994 ms long)
overrun!!! (at least 9.977 ms long)
overrun!!! (at least 7.440 ms long)
overrun!!! (at least 7.608 ms long)
overrun!!! (at least 7.535 ms long)
overrun!!! (at least 8.021 ms long)
overrun!!! (at least 7.242 ms long)
overrun!!! (at least 10.272 ms long)
overrun!!! (at least 9.358 ms long)
overrun!!! (at least 10.346 ms long)
overrun!!! (at least 6.430 ms long)
overrun!!! (at least 7.933 ms long)
overrun!!! (at least 7.630 ms long)
overrun!!! (at least 6.900 ms long)
overrun!!! (at least 7.736 ms long)
overrun!!! (at least 6.608 ms long)
overrun!!! (at least 9.970 ms long)
overrun!!! (at least 6.497 ms long)
overrun!!! (at least 6.948 ms long)
overrun!!! (at least 6.565 ms long)
overrun!!! (at least 9.109 ms long)
overrun!!! (at least 10.157 ms long)
overrun!!! (at least 7.201 ms long)
overrun!!! (at least 7.615 ms long)
overrun!!! (at least 7.468 ms long)
overrun!!! (at least 7.828 ms long)
overrun!!! (at least 7.687 ms long)
overrun!!! (at least 10.449 ms long)
overrun!!! (at least 7.881 ms long)
overrun!!! (at least 7.772 ms long)
overrun!!! (at least 10.053 ms long)
overrun!!! (at least 10.279 ms long)
overrun!!! (at least 7.863 ms long)
overrun!!! (at least 6.796 ms long)
overrun!!! (at least 9.094 ms long)
overrun!!! (at least 10.494 ms long)
overrun!!! (at least 7.640 ms long)
overrun!!! (at least 6.990 ms long)
overrun!!! (at least 10.298 ms long)
overrun!!! (at least 6.823 ms long)
overrun!!! (at least 6.560 ms long)
overrun!!! (at least 7.112 ms long)
overrun!!! (at least 6.477 ms long)
overrun!!! (at least 6.222 ms long)
overrun!!! (at least 6.472 ms long)
overrun!!! (at least 7.637 ms long)
overrun!!! (at least 7.740 ms long)
overrun!!! (at least 7.282 ms long)
overrun!!! (at least 6.445 ms long)
overrun!!! (at least 10.206 ms long)
overrun!!! (at least 6.405 ms long)
overrun!!! (at least 7.439 ms long)
overrun!!! (at least 9.540 ms long)
overrun!!! (at least 7.727 ms long)
overrun!!! (at least 10.262 ms long)
overrun!!! (at least 6.599 ms long)
overrun!!! (at least 7.523 ms long)
overrun!!! (at least 6.789 ms long)
overrun!!! (at least 6.618 ms long)
overrun!!! (at least 7.641 ms long)
overrun!!! (at least 7.376 ms long)
overrun!!! (at least 6.771 ms long)
overrun!!! (at least 7.314 ms long)
overrun!!! (at least 10.433 ms long)
overrun!!! (at least 9.983 ms long)
overrun!!! (at least 9.626 ms long)
Playing back recorded audio...
Did you hear your own voice? (y/n) 

I hope this helps to narrow things down a little bit, and I hope my issue is related to solve two at the same time.

sheridat, 2017-12-17:
@subterain Hi, pigpio causing problems with audio is known about. See the following two posts - 
1. https://github.com/google/aiyprojects-raspbian/issues/197 from someone also using pigpio. If you read it through it becomes obvious what the issue is.
2. https://www.raspberrypi.org/forums/viewtopic.php?f=114&t=199792 from the same poster - in this post he advises that he solved the issue by starting the pigpio daemon as follows sudo pigpiod -t0. You could also amend the service file to start it with that parameter.

subterain, 2017-12-17:
Thanks sheridat, lesson learnt - do a bit more research before posting and reveal my greenness to the world! All working ok now, so thanks again for saving me a chunk of time.

dmitriykovalev, 2019-11-14:
Haven't seen this issue for a long time — closing.

closed 2019-11-14 by dmitriykovalev