google/aiyprojects-raspbian #150: Some commands no longer work after 13/10

shavanni, 2017-10-13:
As of 13/10, some commands no longer work correctly.
Sing me a song now returns "here is information about sing me a song"
Sing a song now returns " here is a matching video"
Tell me a story now returns "here are matching news articles" 

All were working correctly on 12/10

sheridat, 2017-10-13:
Hi, Firstly thanks for giving me some new things to ask it!.
I have just tried those three - I am running the voicekit version of the software.
The first two resulted in it singing me a song. The tell me a story resulted in the opening sentences if a Tale of Two cities. 

How strange. Does the fault replicate after a reboot?

shavanni, 2017-10-13:
No joy I'm afraid. I'm using the gpio trigger, not ok Google but I've done nothing to change it do it seems like a server side problem.

ensonic, 2017-10-13:
If you are using the assistant, maybe you could report this there:
https://github.com/googlesamples/assistant-sdk-python
since we merely integrated this library.

shavanni, 2017-10-14:
I have raised it there.  Apparently it's a known issue

closed 2017-10-14 by shavanni