google/aiyprojects-raspbian #153: BrokenPipeError

Pixel06, 2017-10-15:
Hi, I tried to configure my raspberry pi 3 (with the standard version of raspbian) for the voice kit with _HACKING.md_. The led works but when I run _chek_audio.py_ it returns an error: "BrokenPipeError: [Errno32] Broken pipe". Can you help me, please? Thanks.

t1m0thyj, 2017-10-15:
Have you tried the troubleshooting steps listed [here](https://developers.google.com/assistant/sdk/develop/python/hardware/audio), particularly 2 and 3? It sounds like your Pi is trying to use the wrong playback/recording device.

Pixel06, 2017-10-15:
No, I'm going to try it, thanks

drigz, 2017-10-18:
Good suggestion, @t1m0thyj. @Pixel06, did you try the troubleshooting steps out?

dmitriykovalev, 2018-05-10:
Please reopen if problem still exists.

closed 2018-05-10 by dmitriykovalev