google/aiyprojects-raspbian #155: stopping alarms and timers

yorickvP, 2017-10-18:
Stopping a ringing alarm is very hard right now, you have to time the "ok google" exactly between rings in order to have it recognized. Is there a way to know when alarm is going on and stop it when hitting the button, for example?

drigz, 2017-10-18:
I haven't tested this, but from the [Assistant SDK docs](https://developers.google.com/assistant/sdk/reference/library/python/) it looks like there are events for this. You can try adding `print(event)` to [_process_event](https://github.com/google/aiyprojects-raspbian/blob/0a64297e364c8c2eb8fd2295699c36dcda9ecb3f/src/assistant_library_with_button_demo.py#L69), and then add extra conditions to the script to handle these.

I think this would be too much complexity to add to the example, but if you get it to work, you should follow up on this issue so that others can use it.

closed 2017-10-18 by drigz

yorickvP, 2017-10-18:
The google home can stop timers via hotword. What are they doing differently? Echo cancellation?