google/aiyprojects-raspbian #159: New voicekit branch uses more CPU

t1m0thyj, 2017-10-21:
When I ran the old version of this project on my Pi, the CPU usage indicator in the taskbar showed only around 1% when my Pi was idle. Since updating to the new voicekit branch, the CPU usage has increased and is now always about 9% when idle.

I understand that since the Assistant service is running in the background it needs to use some CPU, but 9% is a significant increase from 1%. I'm not sure what is causing this but think it would be nice if the CPU usage in the new version could be reduced (assuming that it's not just me experiencing this).

divx118, 2017-10-21:
@t1m0thyj You could use `top` on the command line and see which process is using the cpu. I will check  the cpu usage on my rpi 3. Also some more info from you would be welcome, are you running the official voicekit image or did you install it on raspbian or some other distro?

t1m0thyj, 2017-10-21:
Sorry for not providing more info. I am running Assistant on my Pi 2, and am not using the official voicekit image. For both the old branch and the new one, I have been using the Raspbian Stretch 2017-09-07 build, and set up Assistant using the instructions in HACKING.md.

I tried running the `top` command, and it says that the process using the most CPU is the `python3` process running Assistant. The CPU usage it shows stays around 36%, which is 4 times the 9% shown in the taskbar, but I assume that's because the CPU has 4 cores.

mpember, 2017-10-21:
I run the code on a DietPi-based image and the python task idles at 19-20% (or 4.8-5%).

drigz, 2017-10-23:
Nothing comes to mind that would cause this - the new and old versions do pretty much the same thing at their core. We'd need to narrow down the possible causes:

- Upgrading to Stretch
- Possibly different versions of google-assistant-library
- voicekit branch vs master branch
- voicekit branch vs master branch at 48752274 (before the start of the work on the new API)

sheridat, 2017-10-23:
Hi, I have SD cards for both versions 
1. Master branch - From Goggle image - Jessie - CPU utilisation 11-13% python3 CPU 10-11%
2. Voicekit branch - From Google image - Jessie - CPU utilisation 10-14  python3 CPU 10-13%
running on a Pi 3
Both images are up to date with their respective branches
So similar figures.

t1m0thyj, 2017-10-24:
I tested the old master branch on my Pi, both the latest commit and at 4875227, and now it is usually showing a CPU usage of 9-11%. It doesn't seem like the CPU percentage shown in the taskbar is very accurate, because for 30 seconds or so, it was showing 2%, while `top` was showing 36% CPU usage.

It seems like either I am remembering quite wrongly what the situation was before, or something changed in Raspbian (maybe a package update) that affected this. Since I cannot consistently reproduce the 1% CPU usage I mentioned, and 9% isn't an unreasonable amount, I'll close this issue.

closed 2017-10-24 by t1m0thyj