google/aiyprojects-raspbian #160: The speaker and mic have the same levels

xgenius87, 2017-10-22:
Have checked alsamixer and downloaded a GUI for audio levels. Both Input and Output are labeled "Master" and you can't change one without the other doing the exact same thing. 
In the ALSAMixer GUI there is a lock but that lock is useless.
Is there a why to change this? Been spending too much time going in loops it feels.

drigz, 2017-10-23:
Currently the microphone volume can only be adjusted by editing asound.conf: see #47.

It should be possible to adjust the ALSA config to add volume control for the mic: we'd need to keep the amplification from the `route` plugin, and add a `softvol` stage for variable attenuation. We'd also need to make the first stage increase the sample width, to avoid clipping or quantization.

MarsNL, 2017-11-10:
You can use:
amixer set Master 100%
In a terminal screen to change the volume.

For example you can do:
amixer set Master 50%

to set the volume 50 percent.
etc.

dmitriykovalev, 2018-06-25:
PulseAudio handles mic volume changing.

closed 2018-06-25 by dmitriykovalev