google/aiyprojects-raspbian #161: "Do I need an Umbrella" results in -  Ignoring kUnimplemented action error!

sheridat, 2017-10-23:
Hi, I am running the **src/assistant_library_with_local_commands_demo.py**
Whenever I ask it whether I need an umbrella,  I see this 
Say "OK, Google" then speak, or press Ctrl+C to quit...
You said: what day is it
You said: do I need an umbrella
[1546:1558:ERROR:assistant_output_handler.cc(543)] Ignoring kUnimplemented action error!
You said: tell me a joke
You said: do I need an umbrella
[1546:1558:ERROR:assistant_output_handler.cc(543)] Ignoring kUnimplemented action error!
You said: what day is it
You said: do I need an umbrella
[1546:1558:ERROR:assistant_output_handler.cc(543)] Ignoring kUnimplemented action error!

As you can see:
1. It does ignore the "error"
2. The error only appears when I ask if I need an umbrella. 

You might or might not want to investigate.

drigz, 2017-10-23:
Thanks for letting us know.

@Fleker, is this something we should forward to assistant-sdk-python?

Fleker, 2017-10-23:
I think this is a known issue on the assistant sdk side.

@sheridat do you see the same problem when you ask other weather related questions?

sheridat, 2017-10-24:
@Fleker Yes here is an example
Say "OK, Google" then speak, or press Ctrl+C to quit...
You said: will it rain today
[1155:1167:ERROR:assistant_output_handler.cc(543)] Ignoring kUnimplemented action error!
You said: is a cloudy day
[1155:1167:ERROR:assistant_output_handler.cc(543)] Ignoring kUnimplemented action error!

closed 2017-11-14 by sheridat