google/aiyprojects-raspbian #165: High CPU consumption in Rpi3 with "OK Google" trigger enabled

aguedob, 2017-10-31:
When 'OK google' trigger is enabled, the average CPU consumption is around 45%. It this a normal behaviour?

sheridat, 2017-11-02:
No, see [](https://github.com/google/aiyprojects-raspbian/issues/159).
I've just fired mine up using the assistant_library_with_local_commands_demo and it is using according to the Gui Task Manager 12-13% CPU. Mine is running the aiyprojects-2017-09-11.img

drigz, 2017-11-02:
Depending on how you're measuring the CPU usage, it could be the same. Some tools measure as a percentage of one CPU, and some measure as a percentage of all four CPUs. #159 has more info.

@aguedob, how did you measure the CPU usage?

aguedob, 2017-11-02:
Using top command.

sheridat, 2017-11-03:
I measured mine again using top - my cpu usage was 45.3% - I believe my voicekit is working "normally" therefore yes 45% is normal

t1m0thyj, 2017-11-05:
My Pi has the same behavior, as I reported in #159. Based on tests done by @sheridat and me, it seems like the CPU usage is not affected by Raspbian Jessie vs. Stretch, or the old master branch vs. the new voicekit one.

I'm almost certain that when I first used Assistant on my Pi, the CPU usage was significantly lower. My hypothesis is that the update of the google-assistant-library package from v0.0.2 to 0.0.3 is what caused this change, since it's the underlying code that runs the Assistant service, but this is just a guess.

divx118, 2017-11-05:
On my rpi3 with stretch lite, it peeks for a moment to 42%-45% when hearing the "OK Google", but that is only one cpu the other cpu's are idling. When listening for the "OK Google" or in my case also a button press it stays around 30%. When the `status_ui.status` is listening it will drop to almost nothing. Note that I use an adjusted demo with local commands which also supports the button press besides the "OK Google". IMO this behaviour is normal and not too high.
BTW by default `htop` should also be available so you can better see what each cpu is using. That is also what I used to check the cpu usage.

closed 2018-01-08 by drigz